borisbu/OctoRelay #2: Request feature

Syl20-94, 2021-03-16:
Hello, it's works great from scratch ! thank this is what I looked for a while, should it be possible to control it through startup sequence or G-Code command ?
I would like to light on and start enclosure cooling fan when start printing and power them off when print is ended 

thanks

borisbu, 2021-03-22:
It's currently in the develop branch, I'm waiting for a review, and will then release and merge to master

closed 2021-03-22 by borisbu

Syl20-94, 2021-03-22:
I'm looking forward :) thanks a lot; I had a look into octoprint documentation, and I did not find how do it. I will have a look on how did you succeed to implement it.

borisbu, 2021-03-23:
you can install https://github.com/borisbu/OctoRelay/archive/refs/heads/develop.zip in the plugin manager

Syl20-94, 2021-03-23:
Hello, I tried to install this package on a demo and I've got a strange behavior, install is OK I see 1.1.0 package installed, but I lost plugin menu and icons to act, my config.yaml is cleaned of plugin info. 
When I deinstall and install packet from deposit (1.0.0) it running back again.., 

then ...  after enabling debug on the plug in, I log this error 

> 2021-03-23 08:16:38,499 - octoprint.plugin.core - ERROR - Invalid syntax in /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octorelay/__init__.py for plugin octorelay
> Traceback (most recent call last):
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 762, in _parse_metadata
>     root = ast.parse(f.read(), filename=path)
>   File "/usr/lib/python2.7/ast.py", line 37, in parse
>     return compile(source, filename, mode, PyCF_ONLY_AST)
>   File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_octorelay/__init__.py", line 274
>     async def turn_off_pin(self, delay, relay_pin, inverted):
>             ^
> SyntaxError: invalid syntax

digging into net and upgrade my Octoprint to confirm it running Python3 (I through it was the case but I runned the script from https://octoprint.org/blog/2020/09/10/upgrade-to-py3/)
I check an install of async package with "pip install asyncio"
and reinstall everything ... 

and IT WORRRKKSSSS :)

I really like the auto OFF delay idea, I first through it was after the end of print but this is after the ON, maybe it should be inibited when "Auto OFF" is selected to not misleading
Last request if you can implement a delay after the end of printing (should be also for start printing, but not sure of the usefulness)  my usercase is to control a chamber fan that I would like to force heat and particule to goes to filtering 

Good job ! I need to understand how to code it :)

borisbu, 2021-03-23:
I was thinking about that as well, or having a temperature sensor that turns the fan ON if it is to warm.

Syl20-94, 2021-03-23:
OK, I just have to stop discussing, you goes ahead of my comment ;) I'm currently fighting with a DTH22 sensor which refuse to show me temperatur & humidity when I connect it directly on the raspeberry, that why a simple delay should solve my issue for the moment otherwise I will goes to a kind of temperature control for fan speed and I will never stop ;) 
Once more many thanks for your development and fast answer