borisbu/OctoRelay #5: Crash Octoprint after setting pin 37

jiramm, 2021-03-18:
Hi, unfortunately I have a problem setting raley 1. If I set the pin to port 37, it will send the whole octopi to ground. So far, for some unknown reason, the network connection via wifi and Lan will stop working.
After the reboot during the passage 4 pings and immediately the whole network layer dies.
Unfortunately, I tried it on two clean octopi installations with the same result.

borisbu, 2021-03-18:
Hi, you have to set the GPIO number, not the pin, GPIO only goes up to 27.
I'll change the label and limit it to 27
https://github.com/borisbu/OctoRelay/blob/master/img/rpi_gpio.png

closed 2021-03-18 by borisbu

jiramm, 2021-03-18:
OK  Thanks info GPIO NUM 40 confused me