borisbu/OctoRelay #7: does not work

urishani, 2021-03-20:
Hi thanks for that. I really want this one. I use the "Enclosure" plugin (https://github.com/vitormhenrique/OctoPrint-Enclosure) which works for me well, with same pins. Yet, it does not work with your plugin. I tried it with Enclosure disabled with no results.
I think that the options of "Enclosure" are good example for you to add (also "PSU Control"plugin do the same - see https://github.com/kantlivelong/OctoPrint-PSUControl):
1. Option to warn before turning off (important for printer off in the middle of printing)
2. Option to turn off automatically when printing ends, and a time delay in seconds before tha). This is good when you leave printer and want light and fan to stop when not needed anymore.
3. Convert your commands to GET instead of POST and allow to specify if to turn on or off (you just say "update" which I guess is toggle?) That way I can make a URL shortcut on my phone home page to turn on or off in a click (without going through the Octoprint UI). That is what I can do with "Enclosure" [PSU Control uses POST as well unfortunately. :(.  ]

I like that you show different icons on the bar (and not drop down menu as in "Enclosure")
I also like the option to select the icons.
your default 3 relays are exactly what I need too and one plugin instead of a couple is better.
... but it also has to work. ;).

borisbu, 2021-03-22:
Hi, the requested features 1 and two are implemented in the develop branch.
3.) My first try was GET, but did not manage to access the request parameters in the on_api_get() method. But if you want to change something on a server PUT is the right HTTP method.
Why it doesn't work, I don't know. Did you enter the Pins, or the GPIO numbers?

borisbu, 2021-03-25:
Can you please try again?

closed 2021-03-31 by borisbu