borisbu/OctoRelay #12: Installation UnicodeEncodeError error

JustinS-B, 2021-04-02:
Hi, install failed with this error. What can I do to help debug it?

```
2021-04-02 10:25:48,024 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/borisbu/OctoRelay/archive/master.zip
2021-04-02 10:25:48,977 - octoprint.plugin.core - INFO - Plugin Navbar Temperature Plugin (0.14) is disabled.
2021-04-02 10:26:19,776 - octoprint.plugins.pluginmanager - ERROR - Could not install plugin from https://github.com/borisbu/OctoRelay/archive/master.zip
Traceback (most recent call last):
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 738, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1414, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 209, in execute
    return self.call(command, **kwargs)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 511, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 530, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 32: ordinal not in range(256)
2021-04-02 10:26:19,777 - octoprint.plugins.pluginmanager - ERROR - Reason: UnicodeEncodeError('latin-1', '  source/py_gpio.c: In function ‘PyInit__GPIO’:\n', 32, 33, 'ordinal not in range(256)')
Traceback (most recent call last):
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 738, in _command_install_archive
    returncode, stdout, stderr = self._call_pip(pip_args)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/plugins/pluginmanager/__init__.py", line 1414, in _call_pip
    return self._pip_caller.execute(*args, **kwargs)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 209, in execute
    return self.call(command, **kwargs)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 243, in call
    all_stderr += process_stderr(p.stderr.readlines(timeout=0.5))
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 239, in process_stderr
    return process_lines(lines, self._log_stderr)
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 229, in process_lines
    processed = self._preprocess_lines(
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 511, in _preprocess_lines
    return list(map(self._preprocess, lines))
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/pip.py", line 530, in _preprocess
    return to_unicode(clean_ansi(text))
  File "/opt/octoprint-venv/lib/python3.9/site-packages/octoprint/util/commandline.py", line 57, in clean_ansi
    return _ANSI_REGEX.sub(b"", line.encode("latin1")).decode("latin1")
UnicodeEncodeError: 'latin-1' codec can't encode character '\u2018' in position 32: ordinal not in range(256)
2021-04-02 10:26:20,245 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
```

borisbu, 2021-04-02:
looks like you have a character that is not allowed in the config. \u2018 is this here: https://www.fileformat.info/info/unicode/char/2018/index.htm
I'd suggest to delete the plugin data and install again.

JustinS-B, 2021-04-03:
It appears that the problem is 100% with RPi.GPIO and not OctoRelay. I'm not running on a Raspberry OS based system, so the python package RPi.GPIO will never install as there are kernel incompatibilities (see [https://pypi.org/project/RPi.GPIO2/](https://pypi.org/project/RPi.GPIO2/)). All other GPIO packages install normally & run quite happily, and my Bash scripts all turn my relays on and off, so it's not mission critical, it just would have been nice to have run it all through OctoRelay.

```
  ERROR: Failed building wheel for RPi.GPIO
  Running setup.py clean for RPi.GPIO
Successfully built OctoRelay
Failed to build RPi.GPIO
Installing collected packages: RPi.GPIO, OctoRelay
    Running setup.py install for RPi.GPIO ... error
    ERROR: Command errored out with exit status 1:
```
I tried installing RPi.GPIO2 but OctoRelay still tried to install RPi.GPIO and failed in exactly the same place.

borisbu, 2021-04-06:
Hi, what you could do ist run bash scripts instead of switching GPIO pins. I added that in the latest release.

closed 2021-09-17 by borisbu