borisbu/OctoRelay #13: Unable to install

Exalik, 2021-04-03:
Hello,

As I was not able to install it through the octoprint plugin manager, I have tried to install it directly by ssh connection but it does not work anyway and I have:

<blockquote>
pi@octopi:~/oprint$ ./bin/pip install "https://github.com/borisbu/OctoRelay/archive/master.zip"
Collecting https://github.com/borisbu/OctoRelay/archive/master.zip
  Downloading https://github.com/borisbu/OctoRelay/archive/master.zip
     \ 830 kB 1.4 MB/s
Requirement already satisfied: OctoPrint in ./lib/python3.8/site-packages (from OctoRelay==1.1.0) (0+unknown)
Collecting RPi.GPIO
  Downloading RPi.GPIO-0.7.0.tar.gz (30 kB)
Requirement already satisfied: feedparser<7,>=6.0.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (6.0.2)
Requirement already satisfied: zeroconf<0.25,>=0.24 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.24.5)
Requirement already satisfied: pylru<2,>=1.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.2.0)
Requirement already satisfied: OctoPrint-FileCheck>=2020.08.07 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2021.2.23)
Requirement already satisfied: pyserial<4,>=3.4 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (3.5)
Requirement already satisfied: tornado==5.1.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (5.1.1)
Requirement already satisfied: unidecode<0.05,>=0.04.14 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.4.21)
Requirement already satisfied: frozendict<2,>=1.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.2)
Requirement already satisfied: requests<3,>=2.23.0 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2.25.1)
Requirement already satisfied: cachelib<1,>=0.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.1.1)
Requirement already satisfied: Jinja2<3,>=2.11.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2.11.3)
Requirement already satisfied: werkzeug<2,>=1.0.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.0.1)
Requirement already satisfied: flask<2,>=1.1.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.1.2)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.5.0)
Requirement already satisfied: pkginfo<2,>=1.5.0.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.7.0)
Requirement already satisfied: rsa==4.0 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (4.0)
Requirement already satisfied: itsdangerous<2,>=1.1.0 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.1.0)
Requirement already satisfied: blinker<2,>=1.4 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.4)
Requirement already satisfied: Click<8,>=7.1.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (7.1.2)
Requirement already satisfied: sarge==0.1.5post0 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.1.5.post0)
Requirement already satisfied: Flask-Assets<3,>=2.0 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2.0)
Requirement already satisfied: markdown<3.2,>=3.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (3.1.1)
Requirement already satisfied: Flask-Babel<2,>=1.0 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.0.0)
Requirement already satisfied: netaddr<1,>=0.7.19 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.8.0)
Requirement already satisfied: websocket-client<1,>=0.57 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.57.0)
Requirement already satisfied: netifaces<1,>=0.10.9 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.10.9)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2020.09.23 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2021.2.4)
Requirement already satisfied: future<1,>=0.18.2 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.18.2)
Requirement already satisfied: filetype<2,>=1.0.7 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.0.7)
Requirement already satisfied: semantic-version<3,>=2.8.5 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2.8.5)
Requirement already satisfied: watchdog==0.10.4 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.10.4)
Requirement already satisfied: psutil<6,>=5.7 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (5.8.0)
Requirement already satisfied: wrapt<2,>=1.12.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.12.1)
Requirement already satisfied: sentry-sdk<1,>=0.15.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.20.3)
Requirement already satisfied: markupsafe<2.0,>=1.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (1.1.1)
Requirement already satisfied: PyYAML<6,>=5.3.1 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (5.4.1)
Requirement already satisfied: emoji<1,>=0.5.4 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (0.6.0)
Requirement already satisfied: regex!=2018.11.6 in ./lib/python3.8/site-packages (from OctoPrint->OctoRelay==1.1.0) (2020.11.13)
Requirement already satisfied: pyasn1>=0.1.3 in ./lib/python3.8/site-packages (from rsa==4.0->OctoPrint->OctoRelay==1.1.0) (0.4.8)
Requirement already satisfied: pathtools>=0.1.1 in ./lib/python3.8/site-packages (from watchdog==0.10.4->OctoPrint->OctoRelay==1.1.0) (0.1.2)
Requirement already satisfied: sgmllib3k in ./lib/python3.8/site-packages (from feedparser<7,>=6.0.2->OctoPrint->OctoRelay==1.1.0) (1.0.0)
Requirement already satisfied: webassets>=2.0 in ./lib/python3.8/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->OctoRelay==1.1.0) (2.0)
Requirement already satisfied: pytz in ./lib/python3.8/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoRelay==1.1.0) (2021.1)
Requirement already satisfied: Babel>=2.3 in ./lib/python3.8/site-packages (from Flask-Babel<2,>=1.0->OctoPrint->OctoRelay==1.1.0) (2.9.0)
Requirement already satisfied: setuptools>=36 in ./lib/python3.8/site-packages (from markdown<3.2,>=3.1->OctoPrint->OctoRelay==1.1.0) (44.0.0)
Requirement already satisfied: chardet<5,>=3.0.2 in ./lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoRelay==1.1.0) (4.0.0)
Requirement already satisfied: certifi>=2017.4.17 in ./lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoRelay==1.1.0) (2020.12.5)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in ./lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoRelay==1.1.0) (1.26.3)
Requirement already satisfied: idna<3,>=2.5 in ./lib/python3.8/site-packages (from requests<3,>=2.23.0->OctoPrint->OctoRelay==1.1.0) (2.10)
Requirement already satisfied: six in ./lib/python3.8/site-packages (from websocket-client<1,>=0.57->OctoPrint->OctoRelay==1.1.0) (1.15.0)
Requirement already satisfied: ifaddr in ./lib/python3.8/site-packages (from zeroconf<0.25,>=0.24->OctoPrint->OctoRelay==1.1.0) (0.1.7)
Building wheels for collected packages: OctoRelay, RPi.GPIO
  Building wheel for OctoRelay (setup.py) ... done
  Created wheel for OctoRelay: filename=OctoRelay-1.1.0-py2.py3-none-any.whl size=14017 sha256=05e49abd6eea384b178e3e589d4ffb4b223d8f92ba35cdf5a843d0d926ce4a04
  Stored in directory: /tmp/pip-ephem-wheel-cache-g4e1cuoh/wheels/59/0b/25/0e420f64875685ee5ca2be084d1fe9db250950f7b732354da8
  Building wheel for RPi.GPIO (setup.py) ... error
  ERROR: Command errored out with exit status 1:
   command: /home/pi/oprint/bin/python -u -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'/tmp/pip-install-8hk6fuug/rpi-gpio_5e838837998041f8a8bfef873b238950/setup.py'"'"'; __file__='"'"'/tmp/pip-install-8hk6fuug/rpi-gpio_5e838837998041f8a8bfef873b238950/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' bdist_wheel -d /tmp/pip-wheel-2cvur8bq
[... 163 lines cut ...]
    /usr/bin/ld: build/temp.linux-aarch64-3.8/source/constants.o:/tmp/pip-install-8hk6fuug/rpi-gpio_5e838837998041f8a8bfef873b238950/source/constants.h:34: multiple definition of `unknown'; build/temp.linux-aarch64-3.8/sou [... cut ...] mp/pip-install-8hk6fuug/rpi-gpio_5e838837998041f8a8bfef873b238950/setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' install --record /tmp/pip-record-jfk77gra/install-record.txt --single-version-externally-managed --compile --install-headers /home/pi/oprint/include/site/python3.8/RPi.GPIO Check the logs for full command output.
</blockquote>

Could somebody help me please ?

borisbu, 2021-05-01:
can you please try again?

RobinTail, 2023-08-01:
the issue expired

closed 2023-08-01 by RobinTail