borisbu/OctoRelay #14: Not showing new version in plugin update manager

mauricio-camayo, 2021-04-04:
I can see the new version 1.1.0 available in this github, but it is not appearing in the software update manager in octoprint.

[image]

Maybe missing some data as in https://docs.octoprint.org/en/master/bundledplugins/softwareupdate.html#octoprint-plugin-softwareupdate-check-config

mauricio-camayo, 2021-04-05:
Had to uninstall and install again in order to get the new version
[image]
Still missing the "Release Channel" and "Available" version information

mauricio-camayo, 2021-04-06:
I've seen that this repo does not have releases yet https://docs.github.com/en/github/administering-a-repository/managing-releases-in-a-repository 
Maybe it is all that is needed

RobinTail, 2023-07-02:
This issue seems to be fixed at the moment.
In case it's not, please open a new issue or let me know in a comment. 

[image]

closed 2023-07-02 by RobinTail