borisbu/OctoRelay #17: Any way to access the relay state through api?

Deymos42, 2021-04-06:
Hi,

I am looking for the way to get the status of relay sending http request to  octorelay on octoprint server. It's possible?

Thanks in advance.

jneilliii, 2021-12-04:
There appears to be a `listAllStatus` API command added with https://github.com/borisbu/OctoRelay/pull/38.

```
POST /api/plugin/octorelay HTTP/1.1
Host: octopi.local
Content-Type: application/json
X-Api-Key: <your API key>

{
  "command": "listAllStatus"
}
```

Looks like there is a `getStatus` command as well, that appears to need a `pin` parameter, which I think matches up with r1, r2, r3, etc...

```
POST /api/plugin/octorelay HTTP/1.1
Host: octopi.local
Content-Type: application/json
X-Api-Key: <your API key>

{
  "command": "getStatus",
  "pin": "r1"
}
```

RobinTail, 2023-08-01:
[the correct answer](https://github.com/borisbu/OctoRelay/issues/17#issuecomment-986106899) given

closed 2023-08-01 by RobinTail