borisbu/OctoRelay #18: Request - AutoConnect

Lonelyeagledk, 2021-04-08:
Hi

Is it possible to make an automatic connection to the printer from OctoPrint after OctoRelay have powered the printer on. When I power off the printer from OctoRelay the printer is total without power and the connection from OctoPrint are closed.
Thank you :)

Kind Regards
Mads

rich-dobbs-13440, 2021-04-28:
Hi folks!

I just installed the plugin today and have a similar request.  When I reboot Octopi with my current configuration, the printer is powered on, but the connection with the printer has failed.  It succeeds at that point if I click the button to connect.  But it seems like the interface should allow specifying a time to wait before Octopi attempts to connect with the printer.  Let me know if the current architecture of Octopi makes this difficult for you to implement.

Best regards,

Rich Dobbs

Ano10, 2021-05-03:
Hi,
Autoconnect is a good idea. Connect printer after relay is activated would be convenient.
In a same way autodisconnect would be great. Disconnect printer before relay is deactivated would avoid "SerialException" error from octoprint.

Best regards,
Arnaud

RobinTail, 2023-08-01:
That could be really a good feature. I like this idea.

RobinTail, 2023-08-01:
I found out that the following event is being fired after switching the printer's relay on:
```
2023-08-01 22:49:26,133 - octoprint.plugins.octorelay - DEBUG - on_api_command update, parameters {'pin': 'r2', 'command': 'update'}
2023-08-01 22:49:26,230 - octoprint.plugins.octorelay - DEBUG - OctoRelay before pin: 12, inverted: True, relay_state: False
2023-08-01 22:49:27,276 - octoprint.server - INFO - Serial port list was updated, refreshing the port list in the frontend
2023-08-01 22:49:27,289 - octoprint.plugins.octorelay - DEBUG - Got event: ConnectionsAutorefreshed
```

RobinTail, 2023-08-03:
Looks like connection can be implemented similar to EnclosurePlugin

https://github.com/vitormhenrique/OctoPrint-Enclosure/blob/6e35aa59409a71392a341af9a6a54f10cde1df53/octoprint_enclosure/__init__.py#L1789

```py
self._printer.connect()
```

It's not clear when this property or mixin is available.

RobinTail, 2023-08-03:
I checked it, the prop is actually available for the plugin:

```
octoprint.plugins.octorelay - INFO - start OctoRelay
2023-08-03 22:16:03,175 - octoprint.plugins.octorelay - INFO - <octoprint.printer.standard.Printer object at 0x70ee69b8>
2023-08-03 22:16:03,235 - octoprint.plugins.octorelay - INFO - OctoRelay plugin started
```

RobinTail, 2023-08-03:
OK. I also found out that the event I mentioned above is the feature of OctoPrint 1.9.0

https://github.com/OctoPrint/OctoPrint/issues/4473#issuecomment-1318727725

RobinTail, 2023-08-03:
I made a PoC and it works!

RobinTail, 2023-08-03:
> In a same way autodisconnect would be great. Disconnect printer before relay is deactivated would avoid "SerialException" error from octoprint.

This would require for OctoRelay to know which relay controls the printer.
Currently there is no way to figure this out from configuration exception a RexExp against the label, but it's definitely not reliable. I will extract it into another issue once I finish this one, @Ano10

RobinTail, 2023-08-05:
@rich-dobbs-13440 , @Ano10 , @rich-dobbs-13440 ,

should there be a setting to disable/enable AutoConnect?
should there be a delay before connecting?

Let me know what you think.

rich-dobbs-13440, 2023-08-05:
At this point, I have to manually trigger the connection after turning on
the printer by clicking the OctoRelay icon.  The printer starts up and then
the serial port is found.  As soon as I see them refreshed, I click on the
connect button, so there are delays in the process.  I think in practice
you need a delay or some method to detect when the serial port is alive.
For my CR-6 SE, it takes about two seconds between clicking the printer
icon and the serial point being found.

I never click the icon to turn off the printer.  What I do is turn off the
power to my Raspberry Pi with the CanaKit PiSwitch, which shuts off the
printer, since the relay is Normally Closed.  Then I turn the Pi again,
which starts up with the printer off.  Saving the noise from the loud fan
that runs whenever the CR-6 SE is powered on.

In my closer to ideal world, I would just turn the printer on from inside
of Cura.  Though in the longer term I really want to ditch Cura since it is
so poor for automation.

Best regards,

Rich Dobbs

closed 2023-08-05 by RobinTail

rich-dobbs-13440, 2023-08-05:
Anna,

That's great.  When will it be in the release?  If I want to try it before
that, what would I need to do?

Best regards,

Rich Dobbs

RobinTail, 2023-08-06:
Hello @rich-dobbs-13440 ,

the AutoConnect feature is already released in version 3.3.0.
It will only work when running the plugin on OctoPrint 1.9.0 or higher.

You can install OctoRelay 3.3.0 using the following direct link

```
https://github.com/borisbu/OctoRelay/releases/download/3.3.0/release.zip
```

or upgrade your existing one using Software Upgrade section of the OctoRelay settings.

RobinTail, 2023-08-06:
Here is a video showing how the feature works, @rich-dobbs-13440 , @Lonelyeagledk , @Ano10 

https://github.com/borisbu/OctoRelay/assets/13189514/874dd208-f769-4cd1-936d-eb21a32551eb

rich-dobbs-13440, 2023-08-06:
Anna, thanks!  I had already done the upgrades.  I just turned on my
printer using the icon, and it auto connected, without me having to
do anything.  Best regards,  Rich Dobbs

backupartist, 2023-08-16:
I am using a relay switched power strip that physically turns everything off.  I love this feature, however it does not work properly for me. Since I use that power strip and its having to completely reboot, it times out before my printer is fully booted up in order to be detected and connected, Is there a delay setting for this, that I can set before it tries to AutoConnect? That would give my printer time to fully boot up.

PR #149 (2023-08-05): Feat: AutoConnect
Closes #18 

This will only work for OctoPrint 1.9+

I decided to go without additional setting, a possibility of which (in part of its conditional appearance) is being [discussed on OctoPrint community forum](https://community.octoprint.org/t/binding-to-octoprint-version-or-feature-presence/53272/3). And also without any additional or configurable delay. If there will be an issue of a need I will add it later.
PR #152 (2023-08-05): 🚀  3.3.0
Closes #18 

- #149