borisbu/OctoRelay #19: Request - Momentary push button mode

alelucc, 2021-04-10:
Hi,
I'm love with this octoprint addon, it's so intuitive and just works great.
I have a printer with a momentary push button in the front for turning on and off the printer (also has the main switch in the back but it does not turn on the printer) so i was hoping if is possible to add a tick to change the functioning relay mode, between the standard switch and the momentary push button (maybe with a customizable seconds selector).

It would be great, so i can remotely turn on the printer beyond than an external fan and light.

Alessandro

akoirium, 2021-04-15:
hello ... great idea ! ...
i will be interesting for that me too...

but i think the plugin is written only for manage the outputs of the GPIO ... and at this state, not be able to manage inputs ...
But if you are a linux user ... you can write little a script who using wiringPI ( http://wiringpi.com/ ) to implement this function  ... in third party ...

If i have some time ... i will take care of that ... ... 
but the author of the plugin can tell you more than me ...

zanglerecke, 2021-11-29:
i need some similar thing... i want to trigger the Mainboard-Reset. so i need just one second(adjustable?) of High/Low... than it should be reset to the normal position.

also a notification/acc would be great (not only at switch-off direction)

RobinTail, 2023-08-04:
Yes, you can, @alelucc .

I have a similar mechanical switch on my printer. I added a relay after that switch and also connected it to GPIO pin on my Raspberry Pi. 

Depending on the relay, you might also need one of these "logical converters 3-5v":

https://www.amazon.de/-/en/4-Channel-Converter-BiDirectional-Raspberry-Microcontroller/dp/B07RY15XMJ?th=1

Good luck!

closed 2023-08-04 by RobinTail