borisbu/OctoRelay #20: Thermal control ?

akoirium, 2021-04-15:
Sorry, my English is bad (i'm french) and i'm not an advised user of github...

so i post my request here ... i know ... it's not an issue ....

Great plugin ... i love it...
But it is possible to implement a cut off function of a relay who depends of the temperature of the noze or the bed ?....
Like the psu control plugin do but who it's limited to one relay ?
Also ... a little lists of icons in the readme file for the nav bar (what i have to write in the string for each icons ) or a way to modify this more simply ?...

Thanks for you answer ... and also for you plugin who is really great !!!

RobinTail, 2023-08-04:
Regarding switching a relay when some temperature limit has reached — that's clear.
However, you should not rely on OctoRelay plugin as a safety feature — it's not.
By the way, the similar functionality to shutdown the printer in case of overheating should already be available in Marlin firmware.

Regarding icons — you should find you own ones and put html tag `<img src="..." >` into corresponding fields of your settings. You can use their URL from internet or place them among OctoPrint static files.

@akoirium

RobinTail, 2023-08-04:
It can also be the CPU temperature using this implementation:
https://github.com/j7126/OctoPrint-Dashboard/blob/33e73ee470bcee3a5e7409278dabea5b63a2464c/octoprint_dashboard/__init__.py#L147

RobinTail, 2024-12-20:
Ok. Temperature monitoring is out of the plugin scope.
However, some other plugin capable for such monitoring could use the OctoRelay API for switching the relays.

closed 2024-12-20 by RobinTail