borisbu/OctoRelay #26: REQUEST: Shutdown after period off Idle Time

dumbozo, 2021-05-16:
Love the plugin. It is stable, logical and works perfect with my relays.
But... I would like to flip the relays (turn printer and LED off) after a period of idle time.
When the raspi reboots and everything is turned on or when I forget to print or shut it down OctoRelay could switch the relays off.
This would be a very usefull add on to this good plug in!

Thank you in advance.
Grt,
Dumb

RobinTail, 2023-07-31:
The problem here is to define `idle`.
When relay is ON and you don't print anything during a certain amount of time, this could be treated as idle, however, it actually does not mean that as a user you're not doing anything at all. This implies at least reacting to ALL types of events to detect the activity. But even this could be not enough if some sort of UI activity does not produce events on backend.
I'd consider this as a risky feature.

RobinTail, 2023-07-31:
Also, this should not be implemented without that:
https://github.com/borisbu/OctoRelay/issues/87