borisbu/OctoRelay #27: Request: Implementation

Katapippo, 2021-06-02:
Thank you
Great plugin
Is it possible to add the timed shutdown function without starting prints?
It would be very convenient for me to manage the buttons of the Sunlu Dryer Box S1, it starts up via relay but then you have to give 1 click to make it turn on and bring it to temperature, so if I had the possibility to give 1-3 seconds to the relay I could connect it to the button physical to start it remotely.
Did I have a piece of code to add?

Thank you

closed 2023-08-16 by RobinTail

RobinTail, 2023-08-16:
The feature released in v3.8.0 🚀 
@Katapippo

Katapippo, 2023-08-17:
Thank you thank you
You are fantastic if you want you can see this [SHORTS ](https://youtube.com/shorts/hdGF-x0Wdfg?feature=share)how everything works perfectly, Thank you, Thank you

RobinTail, 2023-08-21:
That's great, I'm super glad to read this. Thank you for recording this, @Katapippo 👍🏽

PR #166 (2023-08-16): Feat: duration being turned ON (turn off after turned on)
An additional feature for #28 .
Closes #27 

By setting:
- on Printing Stopped: ON
- after Turned ON: OFF, delay X

User can set the limit for the relay being turned ON.

[image]
PR #183 (2023-08-16): 🚀 3.8.0
Closes #27 and being and addition to #28
Closes #180 

- #182 
- #166