borisbu/OctoRelay #28: Auto ON after print

MelGrubb, 2021-06-12:
I'd like my cooldown fan to turn on for X minutes when a print is completed. There is currently an option to turn relays OFF after a print, but no way to turn them ON.

Sebbabas, 2022-03-20:
hello,
Yes this option would be useful for me too.

mblacklin1, 2022-11-28:
Agreed - would love to see this implemented somehow.

RobinTail, 2023-07-28:
I wanna change settings interface for this, providing a sort of matrix

| Event | Auto ON | Auto OFF |
|--------|:------:|:------:|
| Initially |☑️|  ☑️  |
| Before print | ☑️  | ☑️  |
| After print | ☑️  | ☑️  |

closed 2023-08-10 by RobinTail

RobinTail, 2023-08-10:
@MelGrubb @PalmyreSG1 @Sebbabas @mblacklin1 ,

The feature is available in the recently released version 3.4.0 🚀

RobinTail, 2023-08-16:
>  fan to turn on for X minutes

The feature on turning something on for a limited time released in v3.8.0 🚀 
@MelGrubb

PR #158 (2023-08-10): Feat: internal events and rules
This should make the plugin more flexible for implementing more reactions in the future.
Closes #28 

### Approach

For each relay, instead of having these config props:
```
initial_value, auto_on_before_print, auto_off_after_print, auto_off_delay
```

I'm making a more generic object `rules` having entries matching the following events:
```
STARTUP, PRINTING_STARTED, PRINTING_STOPPED
```

Each of those entries is an object having following properties:
- `state` — the desired relay state in case of the event
  - except `None` meaning "no action needed"
- `delay` [... cut]
PR #165 (2023-08-10): 🚀  3.4.0 
Closes #28 

- #158