borisbu/OctoRelay #29: not there after installation

BobCraig10, 2021-06-28:
I ran installation, it brought up the popup to reboot, but it said installed "unknown" and it does not show up in my list of installed plugins. I tried to install again from zip, it goes through the installation at the end it says, already installed, and closes.

pplisek, 2021-08-17:
Hi, I have the same problem.
2021-08-17 11:37:06,361 - octoprint.plugins.pluginmanager - INFO - Installing plugin from https://github.com/borisbu/OctoRelay/archive/master.zip
2021-08-17 11:37:07,006 - octoprint.plugin.core - INFO - Plugin Pi Support Plugin (2021.3.26.post2) did not pass check, not loading.
2021-08-17 11:37:19,425 - octoprint.plugin.core - ERROR - Error loading plugin octorelay

RobinTail, 2023-08-03:
Please try the newer version using another URL:

```
https://github.com/borisbu/OctoRelay/releases/latest/download/release.zip
```

If the problem remains please open another issue with debug level logging,

@BobCraig10 && @pplisek