borisbu/OctoRelay #31: octoRelay mit TFT Touch

ossilampe, 2021-07-22:
Hallo,

ich habe Octoprint 1.6.1 am laufen, und ein https://www.waveshare.com/5inch-HDMI-LCD.htm dieses belegt die GPIOs

10,  SPI data input of Touch Panel
9,  data output of Touch Panel
25, Touch Panel interrupt, low level while the Touch Panel detects touching
11 SPI clock of Touch Panel
7 Touch Panel chip selection, low active

dennoch funktioniert nach der Installation der Plugin kein Touch mehr auch nicht wenn ich andere GPIOs vergebe.

Relay 1 Aktiv, GPIO 16! aber auch nachj einem neustart funktioniert der Touch nicht mehr
Wenn ich Octorelay Deinstalliere klappt wieder alles..

Bra1nK, 2022-03-23:
I have same problem with an Elecrow 5" touch screen. Touch UI becomes unresponsive when OctoRelay is enabled. Disabling OctoRelay and rebooting the Pi brings back the Touch UI

RobinTail, 2023-08-03:
I've recently fixed a bug in #143 that used to set the initial state of the GPIO (BCM) pins `4`, `17`, `18` and `23` on startup.

It's not clear how it could interfere with @ossilampe 's configuration using `10, 9, 25, 11, 7`, but at least all those side effects eliminated. Please try with at least version 3.2.0.
If the problem remains, enable debug level logging for OctoRelay and open another issue having them attached, @Bra1nK

closed 2023-08-03 by RobinTail