borisbu/OctoRelay #35: icons and labels don't display properly

jacix, 2021-08-05:
Hello,

Thank you for a VERY handy plug-in. My initial setup is for some LEDs mounted below the heatsink fan, to light the print as it's going.  The buttons work like a charm, so the only issue I'm having is that it's ignoring the icon and label settings.  Below is an octoprint.log snippet from when I switched relay 1 on then off, as well as screen shots of my settings page and how OctoRelay looks in the UI, including one of the mouse-over that should read "headlight" instead of "relay 1".  I have two relays enabled in OctoRelay, and both of them work with the UI buttons, though I don't have any lights on the second one yet. I have several other plug-ins that adjust Octo's layout, positioning, etc so I suspect one of those is interfering.  

I'm not a Python expert, but I know my way around so please let me know if there's anything I can do to help.

> 2021-08-05 01:41:16,745 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin octorelay
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 158, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octorelay/__init__.py", line 214, in on_api_command
    self.update_ui()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octorelay/__init__.py", line 308, in update_ui
    GPIO.setup(relay_pin, GPIO.OUT)
ValueError: The channel sent is invalid on a Raspberry Pi
2021-08-05 01:41:18,923 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin octorelay
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/server/api/__init__.py", line 158, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/util/__init__.py", line 1941, in wrapper
    return f(*args, **kwargs)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octorelay/__init__.py", line 214, in on_api_command
    self.update_ui()
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_octorelay/__init__.py", line 308, in update_ui
    GPIO.setup(relay_pin, GPIO.OUT)

<h2>Images</h2>
<h3>UI</h3>
No mouse-over:

[image]

With mouse-over:
[image]

<h3>Settings</h3>

[image]

Pheelix, 2021-09-06:
Try this:
[image]

I don't think it knows to set the correct height or width when using custom images/icons.

langrock, 2022-12-04:
While this post is old, I am having the exact same issue. Installed OctoPrint and configured a single relay. I haven't tried if the relay buttons do anything, but there are four shown on the start page, labeled R1 - R4, while the actual first relay in the config is labeled 'Light', for example.

[image]

Any how this could be fixed?

Thanks

Pheelix, 2022-12-04:
> While this post is old, I am having the exact same issue. Installed OctoPrint and configured a single relay. I haven't tried if the relay buttons do anything, but there are four shown on the start [...]

Do you have R1-R4 Enabled?  Try disabling the ones you are not using.  If you are looking to put a Icon in place of the R#, the best way i have found is to copy a image/icon i want to use to this directory `~/oprint/lib/python3.7/site-packages/octoprint_octorelay/static/img/`  and then use this code for Icon ON/OFF 
Icon ON: `<img highth="20" width="20" src=/plugin/octorelay/static/img/power.png>`
Icon OFF: `<div style="filter: grayscale(100%)">[image]</div>`

langrock, 2022-12-04:
Thanks. All but R1 are disabled, but the first four show up regardless. I'm not too worried about the icons until only the active relays  show up.

RobinTail, 2023-08-03:
This issue is outdated. There were several improvements and fixes to the UI.

Moreover

```
GPIO.setup(relay_pin, GPIO.OUT)
ValueError: The channel sent is invalid on a Raspberry Pi
```

this means either:
- relay pin was not a number and casting it into a number gave a zero
- or you're using wrong pin enumeration. The plugin operates BCM enumeration of GPIO pins.

Please try with the recent version. If the problem remains, please open another issue with debug-level logging.

@jacix && @langrock

closed 2023-08-03 by RobinTail