borisbu/OctoRelay #36: Combining with physical momentary push button

BigJoy75, 2021-09-28:
Hi.. thank you for your plugin.
I have a feature request for this:
I'm using enclosure for my printer and I want to use this plugin to control my air filter fan and enclosure light but I also want to have a physical button to control the air filter fan and the enclosure light. 
For example I can turn on the light using octoprint and turn it off using the physical button and vice versa.

Thank you

skobkars, 2021-09-28:
You can already do it by using two plugins: OctoRelay and Physical Button. That is what I do on my printer.

Cheers,

BigJoy75, 2021-09-28:
> You can already do it by using two plugins: OctoRelay and Physical Button. That is what I do on my printer.
> 
> Cheers,

Would you mind to share on how to do it? 
Do we have to put the same or different GPIO for both plugin?  And what action to put in the Physical button plugin ie. to turn on-off light that has been setup in the Octorelay?

Thanks

skobkars, 2021-09-28:
You can use the same or different GPIO's depending on your setup.

In my case, I have one physical button connected to GPIO3, and GPIO2 is connected to LED lights and configured in OctoRelay, and printer power is controlled by GPIO4 and configured in OctoRelay as well.

As you may know GPIO3 by default is used by Raspberry Pi for waking itself up after shutting down, and OctoRelay is configured to automatically power my printer up on boot, so when my Octopi and printer are off one click of that physical button wakes up Octopi and it in turn switches the printer ON after booting.

Now, when Octopi is running I have Physical Button plugin configured to execute System command "/home/pi/on_reset_button" on that button press, and it contains this script:

```
#!/bin/bash

if [[ "$(gpio -g read 4)" == "1" ]]; then
  gpio -g toggle 2
else
  gpio -g write 4 1
fi

```
As you can see, if the printer is OFF (GPIO4) the button will power it up, but if printer is already ON, it will instead toggle LED lights (GPIO2).

Therefore one button is used for multiple tasks: to wake up Octopi, to power up printer if it was shut down through its own (Klipper) menu or by OctoRelay plugin, and to toggle LED lights, which are also configured in OctoRelay.

So, you can be quite creative with these two plugins.

BigJoy75, 2021-09-28:
Thanks for sharing. I'll try it.

closed 2021-11-05 by BigJoy75