borisbu/OctoRelay #37: How to use "OS Command"

marco-117, 2021-10-17:
Hi, i looked for an explanation for the "OS Command" option.
How can i use them? Is there a list of commands?

Thank you

skobkars, 2021-10-17:
It depends on what are you running your Octoprint/Octopi on. In case of an Octopi, Raspberry Pi is running a [Raspberry Pi OS](https://en.wikipedia.org/wiki/Raspberry_Pi_OS), which is a derivative of Debian Linux, and the user that owns Octopi process is 'pi', so whatever commands that user is capable/allowed to issue, you can use.

In particular, I added a certain script to pi's home directory that manipulates GPIO's and just run that script as a System/OS command.

Cheers,

RobinTail, 2023-07-24:
The nature of OS Commands is well explained above.
I'd like to add that the plugin executes optionally specified OS Commands in event of toggling the pin state.
Meaning that if some relay has just been switched ON or OFF then also an additional OS Command can be executed.

I consider the issue resolved.

closed 2023-07-24 by RobinTail