borisbu/OctoRelay #45: Octorely might cause... webcamd service to stop !

kds69, 2021-12-07:
### What happened:
webcam has stopped to stream since move to 1.7.2 and update plugins.
logs:
2021-12-07 16:25:15,468 - octoprint.plugins.octorelay - INFO - Ocotrelay system command: sudo service webcamd stop

### Versions:
**Octopi 0.17**
**Octorelay 12.0**

### Workaround:
manually start webcamd service: sudo service webcamd start

jneilliii, 2021-12-07:
You probably just need to adjust the relay option for webcam?

kds69, 2021-12-07:
Yep. just found at same time... this is a default config that kciked in which I never set-up myself which has poluted my own set-up when I have upgraded the plugin:
[image]

kds69, 2021-12-07:
it is also quite annoying that this has overriden my fire alert set-up (OFF after printing) !! 
Not sure this was made on purpose to override users original set-up...

RobinTail, 2023-07-23:
This is important UX issue related to the default (initial) settings having:
- activated relays
- connected to some pins that may already be involved into something else
- and even automatic activation (ON when print starts).

Here is the example for `r4`:

```python
            r4=dict(
                active = True,
                relay_pin=23,
                inverted_output=True,
                initial_value=True,
                cmdON="sudo service webcamd start",
                cmdOFF="sudo service webcamd stop",
                iconOn="""<img width="24" height="24" src="/plugin/octorelay/static/img/webcam.svg" >""",
                iconOff="""<img width="24" height="24" src="/plugin/octorelay/static/img/webcam.svg" style="filter: opacity(20%)">""",
                labelText="Webcam",
                confirmOff=False,
                autoONforPrint=True,
                autoOFFforPrint=True,
                autoOffDelay=10,
            ),
```

So, when the user installs the plugin they may not figure out that it's already doing something: toggling relays and executing actual OS commands.

This should be fixed. Settings currently located inside the initial config should become examples or helpers.

RobinTail, 2023-08-02:
In order to fix this issue, we need to maintain config versions first.
It's because OctoPrint stores only the differences from the specified default settings.
So, if I change the defaults and someone is actually using them, it will affect their configuration during the plugin update.

Documentation:

https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.SettingsPlugin.get_settings_version

Example plugin:

https://github.com/Bart1909/OctoPrint-Telegram/blob/master/octoprint_telegram/__init__.py

RobinTail, 2023-08-02:
But probably not. It's not clear from the documentation how smart this thing is.
Perhaps I should conduct an experiment on this.

jneilliii, 2023-08-02:
This might help. https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/blob/06006ed6710bab173d8000ed916bbab483314187/octoprint_tplinksmartplug/__init__.py#L220

RobinTail, 2023-08-03:
Thank you, @jneilliii, I tried to make an implementation similar to the one you mentioned.
I also made a system for running migrations in a queue.

Thank you, @kds69, for reporting the issue.
The fix will be releasing in v3.2.0 soon.

PR #143 (2023-08-03): Disabling all relays by default.
Closes #45 

This PR fixes an issue, but it also introduces a feature — settings versioning and migrations.
PR #144 (2023-08-03): 🚀 Releasing 3.2.0
Closes #45 

- #143
closed 2023-08-03 by RobinTail