borisbu/OctoRelay #46: Resets when print starts

GordonMKF, 2021-12-20:
Setup relay to switch printer off and on. Works fine until print loaded and starts then printer is shutoff.
Setup a relay on a second Pi with another printer but only controlling a light and same situation. On print start light switches off. Print continues and I can switch the light on without any issues.

[image]

Printer on/off relay is externally powered so only signal triggering. Light relay on second Pi is powered from Pi.
Raspi 3B+'s with all updates done.

mArJaNu-yV, 2022-01-27:
Same situation. orctoprint 1.7.3.

NDR008, 2022-02-08:
I got the same issue... :/

borisbu, 2022-02-09:
For my relay board, I need to invert the output. Did you try to do that?

[image]

NDR008, 2022-02-09:
@borisbu I am using a different module that was design for the ESP D1 mini.
But basically your program is working for turning on / off when I click the icon.
It just turns off when I click print.

Question - is the GPIO being used in pull-down or pull-up? Are you setting the internal resistor?

NDR008, 2022-02-09:
@borisbu I understood what you mean... if you use the relay the opposite way, drive the relay to keep the printer off, the bug does not occur... but there must be a reason for this.
Let me try to poke at the code myself :)

NDR008, 2022-02-09:
@borisbu Update again: If I edit this line:
https://github.com/borisbu/OctoRelay/blob/5a940f5fa253930341db3ae0e864bb691fb39a5e/octoprint_octorelay/__init__.py#L320
From
GPIO.output(relay_pin, inverted != True)

to
GPIO.output(relay_pin, inverted == True)
The relay works as expected without Inverted settings
So somewhere here something is bizarre...

NDR008, 2022-02-09:
I made a pull request that should fix it.

Diaprojector, 2022-03-03:
Hi, I've had this issue as well (came here to report it). Seems like this issue is only on the GPIO23 pin. Changed it to GPIO17 and had no issues.

jshep321, 2022-04-30:
Hi,
A fresh install of octoprint on a new pi4 using a powerswitch tail relay (https://www.adafruit.com/product/268).  GPIO18 and gnd connected to the relay.  OctoRelay works perfectly until I actually start a print, then clicks off.  Non-inverted mode.

I moved over to the PSU Control plugin, (which uses the deprecated GPIO library I believe) and it works fine.

zlomennypez, 2022-05-20:
This error has the source in printer icon settings. If you set the printer relay to light, it works normally.

jamesturner-mark43, 2022-06-30:
> @borisbu Update again: If I edit this line:
>
> https://github.com/borisbu/OctoRelay/blob/5a940f5fa253930341db3ae0e864bb691fb39a5e/octoprint_octorelay/__init__.py#L320
>
>
> From
> GPIO.output(relay_pin, [...]

Can confirm that this fix worked for me.

hcooper, 2022-12-02:
I'm going to suggest this is related to what I fix in https://github.com/borisbu/OctoRelay/pull/55 - the inactive relay configs are being processes like they're active, and one of the default configs conflicts with the relay you've setup.

See my suggested quick fixes here:
 - https://github.com/borisbu/OctoRelay/issues/39#issuecomment-1329686757
 - https://github.com/borisbu/OctoRelay/issues/52#issuecomment-1329695190

closed 2023-07-23 by RobinTail

RobinTail, 2023-07-23:
🚀  The issue most likely fixed in 2.0.1.
In case it's not, please create another one, @GordonMKF

PR #88 (2023-07-23): Releasing 2.0.1
- #48 

Closes #46