borisbu/OctoRelay #50: Orange pi and Octorelay

Antxon35, 2022-04-23:
I am trying to make Octorelay work with an OrangePi instead of a Raspberry Pi.

so far i managed to get some kind of answer but definitely not working properly and I am running out of ideas.

The modificaitons I did to the __init__.py file where:

import orangepi.winplus library (definition of my board GPIO pinout)
change from RPi to OPi library
use orangepiwinplus.BOARD mode for the pin mapping of my specific model.

I defined a pin to be an output (my intention is to have a light that I can swicht on and off) and when the system boots I can see the output is activated but I cannot change it later on octoprint.

In Octoprint log I keep having this error:

_2022-04-23 12:13:32,723 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin octorelay
Traceback (most recent call last):
  File "/home/itu/OctoPrint/venv/lib/python3.8/site-packages/octoprint/server/api/__init__.py", line 158, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/itu/OctoPrint/venv/lib/python3.8/site-packages/octoprint/util/__init__.py", line 1737, in wrapper
    return f(*args, **kwargs)
  File "/home/itu/OctoPrint/venv/lib/python3.8/site-packages/octoprint_octorelay/__init__.py", line 245, in on_api_command
    GPIO.setup(relay_pin, GPIO.OUT)
  File "/home/itu/OctoPrint/venv/lib/python3.8/site-packages/OPi/GPIO.py", line 469, in setup
    raise RuntimeError("Channel {0} is already configured".format(channel))
RuntimeError: Channel 37 is already configured_
[image]

relay icons are not shown either

[image]

Using python console I am able to change the state of the pin without problems.

Any idea how to make it work?

RobinTail, 2023-08-04:
> RuntimeError: Channel 37 is already configured

you need to add 

```py
GPIO.setwarnings(False)
```

before the faulty statement, @Antxon35

RobinTail, 2023-08-04:
> relay icons are not shown either

that's because python execution does not get to `update_ui()` method.
if you fix python, perhaps it will work.
or there is also a JS error (check the browser console).

RobinTail, 2023-08-06:
@Antxon35 ,
consider trying your fork on the base of the most recent version of OctoRelay.
I made a file `driver.py` having all the GPIO operations in a single place.

closed 2023-08-06 by RobinTail