borisbu/OctoRelay #51: Control Buttons Not Being Hidden From Read Only Users

plazarch, 2022-04-28:
When logging in as a read only user the OctoRelay icons still show in the status bar and I am able to toggle the light.  I have not tried toggling the printer power as I am in the middle of a print, but I would assume it would power off the printer also.  It appears the plugin does not adhear to the access groups set up in OctoPrint.  Please advise / fix.  I am on OctoRelay v1.2.0 and OctoPrint v1.7.3  Let me know if you need any other info.

RobinTail, 2023-07-23:
This is actually important and should be fixed.

RobinTail, 2023-07-24:
Could be useful later: https://docs.octoprint.org/en/master/plugins/mixins.html#octoprint.plugin.SimpleApiPlugin.is_api_adminonly

RobinTail, 2023-07-25:
This one we can use to set up a custom permission on controlling the relays:
https://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-access-permissions

RobinTail, 2023-07-25:
@plazarch ,

I figured out how to prohibit switching the relay state for readonly users in #99 .
However, it's not clear for me what should happen to relays having "ON initially on boot" and "Auto ON/OFF before/after printing" for that user.

What do you think?

RobinTail, 2023-07-26:
I decided to go with option 2 and prohibit automatic toggling as well.
For security reasons.

RobinTail, 2023-07-26:
And this was a wrong decision :)
Those events are happening outside the user context.

plazarch, 2023-07-26:
That's awesome.  Thank you very much!

PR #99 (2023-07-26): Prohibit operating relays for readonly users
Closes #51 

Approach: introducing a custom permission.

- Initially enabled for Admins and Users (Operators).

Constraints:

- Avoid showing the buttons
- Reject the API command `update`
- Avoid toggling relays having "ON initially on boot"
- Avoid toggling relays having "Auto ON/OFF before/after printing"
PR #100 (2023-07-26): Relay switching permission
Closes #51 
Continuation of #99 

It turned out that `on_` events are happening outside the user context, so they don't have access to Permission.
PR #101 (2023-07-26): Releasing 2.1.0
Closes #51 

- #100