borisbu/OctoRelay #52: ON before printing is "stuck" activated, despite NOT being activated in settings!

IAmOrion, 2022-05-25:
[image]

As you can see in my screenshot, "ON before printing" is _NOT_ ticked/active.  Yet, every time I start a print, the relay turns on! 
With my relay board, the inverted output is required.  For all intents and purposes it works as expected if I manually toggle the relay control.  It also correctly turns OFF when a print is finished as expected.  I just don't want it turning on when a print starts... but for some reason it does.

shooter-z, 2022-07-31:
Can confirm. Octorelay 1.2.0 running on raspberry pi.
It is not a display bug, since behaviour is the same whenever "ON before printing" is enabled or disabled.

IAmOrion, 2022-11-23:
I'm assuming this plugin is dead/no longer supported. :(

hcooper, 2022-11-28:
I've not tested your exact problem, however I fixed an issue with "on before printing" not working for me.

https://github.com/borisbu/OctoRelay/pull/55

The issue was that the `print_started()` function, was processing all relay configs, not just the "enabled ones". At but chance the default configs were trying to use the same pin as me.

From your screenshot I see you're using pin 23. That's the same pin the default config for Relay 4 uses. Until my PR is committed, try changing Relay 4 config to a different/fake pin and see if that fixes your problem.

IAmOrion, 2022-12-01:
@hcooper - well done discovering that! That fixed the problem for me, thanks for the info.

RobinTail, 2023-07-19:
🚀  @IAmOrion , the issue should be fixed in v1.4.1.
Please check it out, and in case it's not, please create another issue.

PR #77 (2023-07-19): Releasing `1.4.1`
- #55 

Fixes #39 
Fixes #52
closed 2023-07-19 by RobinTail