borisbu/OctoRelay #53: Cannot install on OctoCritrico (Armbian)

Rick88Stang, 2022-08-31:
Just getting Installed: Unknown when trying to install OctoRelay, after reboot it does not show up. It seems to install but cannot find it in plugins. Here is a screen shot and the install process.

[image]

`Installing plugin "OctoRelay" from https://github.com/borisbu/OctoRelay/archive/master.zip...
/home/pi/oprint/bin/python3 -m pip --disable-pip-version-check install file:///tmp/tmp8auxnaex/OctoRelay-master.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmp8auxnaex/OctoRelay-master.zip
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.7/site-packages (from OctoRelay==1.2.0) (1.8.2)
Requirement already satisfied: RPi.GPIO in /home/pi/oprint/lib/python3.7/site-packages (from OctoRelay==1.2.0) (0.7.1)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2021.2.23)
Requirement already satisfied: flask<2.2,>=2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.1.3)
Requirement already satisfied: markdown<4,>=3.2.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (3.4.1)
Requirement already satisfied: sentry-sdk<2,>=1.5.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.9.5)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.8.0)
Requirement already satisfied: frozendict<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.3.4)
Requirement already satisfied: watchdog<2,>=1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.0.2)
Requirement already satisfied: Flask-Babel<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.0.0)
Requirement already satisfied: werkzeug<2.1,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.0.3)
Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.4.21)
Requirement already satisfied: emoji<2,>=1.4.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.7.0)
Requirement already satisfied: pylru<2,>=1.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.2.0)
Requirement already satisfied: tornado<7,>=6.0.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (6.2)
Requirement already satisfied: filetype<2,>=1.0.7 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.0.7)
Requirement already satisfied: regex in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2020.11.13)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2021.10.11)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.0)
Requirement already satisfied: psutil<6,>=5.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (5.8.0)
Requirement already satisfied: Click<9,>=8.0.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (8.1.3)
Requirement already satisfied: pyserial<4,>=3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (3.5)
Requirement already satisfied: PyYAML<6,>=5.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (5.4.1)
Requirement already satisfied: requests<3,>=2.26.0 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.28.1)
Requirement already satisfied: future<1,>=0.18.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.18.2)
Requirement already satisfied: zeroconf<0.34,>=0.33 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.33.4)
Requirement already satisfied: sarge==0.1.6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.1.6)
Requirement already satisfied: pkginfo<2,>=1.7.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.8.3)
Requirement already satisfied: Flask-Login<0.6,>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.5.0)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.3.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.3.4)
Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.11.0)
Requirement already satisfied: semantic-version<3,>=2.8.5 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.8.5)
Requirement already satisfied: feedparser<7,>=6.0.8 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (6.0.10)
Requirement already satisfied: cachelib<0.3,>=0.2 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (0.2.0)
Requirement already satisfied: blinker<2,>=1.4 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.4)
Requirement already satisfied: websocket-client<2,>=1.2.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.4.0)
Requirement already satisfied: colorlog<7,>=6 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (6.7.0)
Requirement already satisfied: OctoPrint-PiSupport>=2022.3.28 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2022.6.13)
Requirement already satisfied: pathvalidate<3,>=2.4.1 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (2.5.2)
Requirement already satisfied: wrapt<1.14,>=1.13.3 in /home/pi/oprint/lib/python3.7/site-packages (from OctoPrint->OctoRelay==1.2.0) (1.13.3)
Requirement already satisfied: importlib-metadata in /home/pi/oprint/lib/python3.7/site-packages (from Click<9,>=8.0.3->OctoPrint->OctoRelay==1.2.0) (4.12.0)
Requirement already satisfied: sgmllib3k in /home/pi/oprint/lib/python3.7/site-packages (from feedparser<7,>=6.0.8->OctoPrint->OctoRelay==1.2.0) (1.0.0)
Requirement already satisfied: Jinja2>=3.0 in /home/pi/oprint/lib/python3.7/site-packages (from flask<2.2,>=2.1->OctoPrint->OctoRelay==1.2.0) (3.1.2)
Requirement already satisfied: itsdangerous>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from flask<2.2,>=2.1->OctoPrint->OctoRelay==1.2.0) (2.1.2)
Requirement already satisfied: webassets>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Assets<3,>=2.0->OctoPrint->OctoRelay==1.2.0) (2.0)
Requirement already satisfied: Babel>=2.3 in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<3,>=2.0->OctoPrint->OctoRelay==1.2.0) (2.9.0)
Requirement already satisfied: pytz in /home/pi/oprint/lib/python3.7/site-packages (from Flask-Babel<3,>=2.0->OctoPrint->OctoRelay==1.2.0) (2020.5)
Requirement already satisfied: typing-extensions>=3.6.4 in /home/pi/oprint/lib/python3.7/site-packages (from importlib-metadata->Click<9,>=8.0.3->OctoPrint->OctoRelay==1.2.0) (4.3.0)
Requirement already satisfied: zipp>=0.5 in /home/pi/oprint/lib/python3.7/site-packages (from importlib-metadata->Click<9,>=8.0.3->OctoPrint->OctoRelay==1.2.0) (3.8.1)
Requirement already satisfied: MarkupSafe>=2.0 in /home/pi/oprint/lib/python3.7/site-packages (from Jinja2>=3.0->flask<2.2,>=2.1->OctoPrint->OctoRelay==1.2.0) (2.1.1)
Requirement already satisfied: charset-normalizer<3,>=2 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.26.0->OctoPrint->OctoRelay==1.2.0) (2.1.1)
Requirement already satisfied: certifi>=2017.4.17 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.26.0->OctoPrint->OctoRelay==1.2.0) (2020.12.5)
Requirement already satisfied: idna<4,>=2.5 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.26.0->OctoPrint->OctoRelay==1.2.0) (2.10)
Requirement already satisfied: urllib3<1.27,>=1.21.1 in /home/pi/oprint/lib/python3.7/site-packages (from requests<3,>=2.26.0->OctoPrint->OctoRelay==1.2.0) (1.26.12)
[... cut ...]
Created wheel for OctoRelay: filename=OctoRelay-1.2.0-py2.py3-none-any.whl size=38623 sha256=cd615b30d808953381cce4edd817e605b06729cc5940037026a45c328917ccd4
Stored in directory: /tmp/pip-ephem-wheel-cache-al8jrnwk/wheels/80/7c/78/128122a64b60ee972a778fd5dd96e98bac872b0d4a227dbf64
Successfully built OctoRelay
Installing collected packages: OctoRelay
Attempting uninstall: OctoRelay
Found existing installation: OctoRelay 1.2.0
Uninstalling OctoRelay-1.2.0:
Successfully uninstalled OctoRelay-1.2.0
Successfully installed OctoRelay-1.2.0
Done!`

Thank you for any help

jneilliii, 2022-08-31:
typically the unknown plugin name indicates that a dependency is failing. I suspect this is probably coming from the RPi.GPIO library, as I don't believe that is compatible with armbian devices. The plugin would have to be re-programmed using a different GPIO module, like `python-periphery` (PSU Control made this transition) to be compatible I think.

Rick88Stang, 2022-08-31:
Got ya! Ok that would explain it! Thank you! I'll find some other way. Unfortunately my Raspberry pi died on me so I'm stuck with this for now

jneilliii, 2022-08-31:
You could technically use PSU Control plugin to manage one relay with simple on/off.

Rick88Stang, 2022-08-31:
I'll look into it. I need to run a few relays, enclosure lights, and fans. I may have found something that works but I'll have to test it and get back to yall.

RobinTail, 2023-08-04:
The issue [resolved](https://github.com/borisbu/OctoRelay/issues/53#issuecomment-1233106797)

closed 2023-08-04 by RobinTail