borisbu/OctoRelay #54: Gcode Support

manmac2, 2022-10-21:
Hi,
thanks for this great plugin. I am wondering if you could add an option to send also a GCODE command on relay on/off.
e.g. to unload filament via M702.

Thanks
manmac

RobinTail, 2023-08-17:
The problem with that is to avoid a potential infinite loop, since the plugin also listens to `@OCTORELAY` command in GCODE.
Also, GCODE commands can only be sent when the connection to the printer is established.
At the moment, plugin is not aware of the connection state, however recently I made an Autoconnect feature #149 requiring OctoPrint 1.9+, that could help with this request.

RobinTail, 2024-04-30:
Featured in 2.2.0
https://github.com/borisbu/OctoRelay/blob/master/CHANGELOG.md#220

closed 2024-04-30 by RobinTail