borisbu/OctoRelay #56: Labels and relays not working on OctoPrint 1.8.6 / Python 3.9.2

langrock, 2022-12-06:
I installed OctoPrint on a Raspberry Pi 2B running DietPi. Everything works quite well, but I cannot get OctoRelay to work at all. After installation, and deactivating all but the first relay entry, the UI still shows relays 'R1 R2 R3 R4' on the dashboard. The labels are being ignored, the icons are ignored, and attached relays are not being actuated when pressing the links. I tested an attached relay using the RPi.GPO module and it works just fine. It's attached to GPIO 4 (pin 7), which is also what the default setting for relay 1 under OctoRelay uses. Any way to troubleshoot this? I already removed and re-installed OctoRelay, but nothing changed. Are there any dependencies that may need to be fulfilled? It'd be nice to get this to work.

Thanks

kroshira, 2022-12-08:
@langrock im not one of the devs but i did run into this issue as well. i am running octoprint in a docker container. the solution for me was making the docker container run as privileged. might be something similar for you

langrock, 2022-12-08:
That's a good tip, thanks. While I'm not running the app inside a container, I should check as which user it is running and then make sure that the user is part of the gpio group. On the other hand, this would only explain why the relay buttons don't do anything, but not why the labels don't work or why de-activated relay entries show up on the dashboard.

kroshira, 2022-12-08:
as soon as i gave the privileged permissions to the docker container the labels showed up and the inactive relay buttons disappeared. i think there is likely an exception that is being raised prior to instantiating the correct active buttons and labels.

langrock, 2022-12-09:
That's good to know. Would be great if it worked. Thanks again.

langrock, 2022-12-09:
This indeed solved the problem for me as well. I added the 'octoprint' user to the gpio group and rebooted the Pi. The relay works, and I only see the activated relays in the toolbar as long as those have an icon associated with them. Thanks a lot for pointing me into this direction.

RobinTail, 2023-07-04:
The issue resolved

https://github.com/borisbu/OctoRelay/issues/56#issuecomment-1343856916

closed 2023-07-04 by RobinTail