borisbu/OctoRelay #57: Auto-On does not seem to work with BigTreeTech SKR Mini E3

MartinVinter, 2023-01-31:
After having switched the mainboard of my ender 3 to the one mentioned above, the feature "ON before printing" does not seem to cause the GPIO pin to trigger.

I can still switch the same pin on manually by pressing the right button in the  UI. That is how I know it is still working.

The main difference between the BTT board and the original Creality board seems to be that the creality board was still powered through USB even when the printer was off. The BTT board appears to be completely off when the power for the printer is cut.

RobinTail, 2023-08-04:
Hello, @MartinVinter 

- In the "Logging" section of OctoPrint please enable "debug" level logging for OctoRelay
- restart OctoPrint
- try printing
- then proceed to the logging section again and download the log of OctoPrint.

There should be some information on the events and actions OctoRelay performed. Please share the corresponding portion.
Also consider upgrading to the most recent version of the plugin since several bugs were already fixed.

RobinTail, 2024-04-30:
stale

closed 2024-04-30 by RobinTail