borisbu/OctoRelay #58: Stop the Smart Filament Sensor plugin

Sc0ut71, 2023-02-23:
When I install the plugin it stops me from detecting filament via the Smart Filament Sensor plugin. As soon as I uninstall everything works fine again. If I configure the relay board with the Enclosure plugin everything works fine. I'd like to use this plugin because it has all the functions I'm looking for but the filament sensor is essential for me... are there any solutions?

RobinTail, 2023-07-21:
- - @Sc0ut71 , in case that filament sensor is attached to a GPIO pin, than it's possible that it was interfering with a configuration of deactivated relays in OctoRelay. This issue was recently fixed in #55. The fix was released in v1.4.1. 
Please retry with a recent version of OctoRelay and let me know the outcome. If it's still failing, try to collect more information, for instance from logs of both plugins.

closed 2023-08-03 by RobinTail

RobinTail, 2023-08-03:
The conflict may also come from using Board mode pin enumeration in the filament sensor plugin (which is default one, I presume).

OctoRelay relies on BCM mode of pins enumeration.

Since both plugins operate using the same `RPi.GPIO` library, perhaps they can not work simultaneously having different modes of pin enumeration.

CC @Sc0ut71 && @pino7421