borisbu/OctoRelay #59: [HELP] how os commands with multiple instances

hareyleyland, 2023-02-25:
i want to be able to control the relays connected to a raspberry pi octoprint instance from a pc octoprint instance via api so that i can control the same relay from multiple instances as i have 2 printers sharing lights fans etc etc how can i implement this maybe with the os command to send a api post telling the relay to switch on? it seems you already added a feature that checks if gpio state changes in background and to update ui so it seems like this would be a logical feature implementation. do you think remote gpio would work? which i could then use a os command?

RobinTail, 2023-08-04:
The plugin shows the relay state regardless a computer you operate the UI from.
Please try with the recent version of the plugin and if something does not work, please provide more details on exact issues.
@hareyleyland

RobinTail, 2024-04-30:
stale

closed 2024-04-30 by RobinTail