borisbu/OctoRelay #60: Add physical button control

Randomaker, 2023-03-07:
It would be amazing if this plugin had functionality to turn relays on/off with a physical button connected to a GPIO pin. For example, if I want to change filament quick, I don't really want to have to get my phone/PC out and turn my printer on, it would be great if I could just add a physical on/off button.

RobinTail, 2023-07-18:
How exactly the plugin could help you with that?

The plugin is capable to reflect the pin state. When you toggle the pin state externally (outside the OctoPrint), you will see that the corresponding control icon has changed.

Therefore, I don't see an obstacle that could block you from creating a physical button operating the same pin as the one controlled by the plugin. Thus, you'd have two way of toggling it.

Otherwise, please provide more details on what in particular is missing in the plugin or blocking you from that desire, @Randomaker

RobinTail, 2023-07-28:
Not clarified.
Considered as not an issue.
In case it's wrong, please open another one and provide more details.

closed 2023-07-28 by RobinTail