borisbu/OctoRelay #61: Relays with time program

quantenquark, 2023-03-10:
Hi, 
I am using the plugin which is working perfectly. Thanks a lot for your work!

Would it be feasable to implement a time program for one channel as an additional feature to allow intermittent switching operations?

Best
Achim

RobinTail, 2023-08-04:
Hello @quantenquark 

Please provide an example of such a time program and switching operation.

quantenquark, 2023-08-04:
Hi Robin, I could imagine a kind of timer to keep e.g. fans or an exhaust system in operation for another 5 minutes after printing.

Regards
Achim

RobinTail, 2023-08-05:
> Hi Robin, I could imagine a kind of timer to keep e.g. fans or an exhaust system in operation for another 5 minutes after printing.

Hello @quantenquark ,

The plugin already has these two settings for each relay:

- OFF after printing	— Turns the relay OFF after finished printing
- Delay	— Postpones turning the relay OFF for X seconds

Would they be enough to achieve the desired behavior you mentioned?

Otherwise what is missing?

RobinTail, 2023-08-05:
Or you meant that you want to turn ON something after print and after some time turn it OFF?

quantenquark, 2023-08-05:
Definitly off but selecting the state World also ve a benefit.

RobinTail, 2023-08-17:
> I could imagine a kind of timer to keep e.g. fans or an exhaust system in operation for another 5 minutes after printing.

Implemented by #166 , released yesterday in v3.8.0 🚀 , @quantenquark

closed 2023-08-17 by RobinTail