borisbu/OctoRelay #62: OctoRelay stoped working after updating octoprint to Version 1.8.7

spawnreaper, 2023-03-23:
I just updated to Version 1.8.7 and noticed that the printer no longer turns off after the print is done. De Octorelay buttons stoped functioning. Checked if my settings have changed but all is the same af before. Noting changed in the hardware setup.

reivaxy, 2023-04-07:
Hi, I have 1.8.7 and I also can't make this plugin work :(

hcooper, 2023-05-02:
Sorry, not very helpful, but just to note I've just done a fresh install of octopi (which installed octoprint 1.8.7) and can confirm OctoRelay does work ok for me (I came back to github to find the minor fix I explain in #55).

RobinTail, 2023-07-23:
Hello @spawnreaper ,

- There were several fixes recently applied to the plugin (those could be the reason)
- OctoPrint 1.9 with its own fixes has been released.
- I [tested](https://github.com/borisbu/OctoRelay/pull/90) the plugin both against OctoPrint 1.8 and 1.9 and different Python versions.
- There is not enough information in your report to figure out what has happen.

Please retry with:
- OctoPrint 1.9.2
- OctoRelay 2.x

and if the issue remains — create another issue.
Please also try to collect logs from OctoPrint, OctoRelay and the console of your browser.

closed 2023-07-23 by RobinTail