borisbu/OctoRelay #87: (Feature Request)  Off after printing delay cancel button

backupartist, 2023-07-22:
[image]

First of all I have to say, I love this plugin, I truly use it everyday, all the time. Thank you for maintaining it.  I have my setup on a relay controlled power strip, that as you can see in the picture, after the print is finished, basically turns everything off after 300 seconds, (5 Minutes).   I can just leave a print going, and when its done everything is turned off. It works amazing, I don't have to worry about it all being left on, after a long print that I may leave running as I go to work or whatever. When its done printing, after 5 minutes of fan cool down it all shuts off.  What I would like to see, is a cancel button for that timer, to cancel the process so it does not turn the relay off. As it stands, since I have it set for 5 minutes, if I don't have another print ready to go, and start printing within that time, It shuts the relay off, which in turn shuts my printer, lights, fans and everything off (which it is supposed to). I  then have to manually turn it back on and wait for it to reboot my printer, and the reconnect with octoprint. Its not terrible, but it gets annoying after you have to do it a few hundred times.  If there was maybe a little button that just says cancel, that pops up next to the printer on and off icon, or whatever relay your particular setup is using, that would allow to cancel the "OFF after printing delay" That would be amazing. Or if it maybe had a countdown time with the seconds left before it shuts the relay off in the button so you know how much time is left to cancel it. That would be awesome. I don't know enough python to attempt to try and do it myself.

Sorry if I didn't post this in the right place, this is my first time ever posting on here, I actually just signed up just to ask this. 

Again, thank you for maintaining this awesome plugin, and keep up all the good work.

RobinTail, 2023-07-23:
I like this idea.
The problem you mentioned is an actual UX issue and the reason why I don't use `Auto OFF`.
I'd like to implement this, but so far I have no yet and idea what the UI control for cancelling should look like.

Also, it's important to mention, that 3D printers are potentially dangerous devices that may cause fire being left unattended.
So, you should never rely on `Auto OFF` as a "safety feature", it's not.

I will take a look on what I could come up with for this feature.
Meanwhile, all contributions and ideas, including UI sketches are welcome.
Thank you, @backupartist .

RobinTail, 2023-07-23:
Perhaps we could use a notification style above the control to indicate that it's being on timer.
In this case, clicking on the control will disable the timer.

[image]

OR

Maybe it's better showing an additional hint-like button without any interference with the original control

[image]

backupartist, 2023-07-24:
[image]

I like this one, I think it is a beautiful design, clean and intuitive. It would be perfect if it could be implemented like that. I just have one other addition to it, if its possible, as I don't know how much coding it would take.   The time left countdown timer that shows how many seconds are left till it shuts off the relay. I did a quick edit with the numbers in red as an example. That way you know how fast you have to click the cancel button. I think it would be handy,  however if its a lot of coding then I can completely understand.  But this implementation either way, would be amazingly awesome.

backupartist, 2023-07-24:
As a quick side note, I agree the auto off feature, as a safety feature should not be solely relied on by myself or others. As we all know, these things can at times spontaneously combust.  I do however have mine in a fire retardant enclosure, that is also vented, I do a lot of nylon and ABS. It is also out in my shop away from most flammables, so I feel a little safer. I do not solely recommend relying on the auto off as a safety feature still however.  So on a liability note, use it carefully.

jneilliii, 2023-07-24:
It's not as pretty, but I do a count-down timer in my IoT power plugins (TPLinkSmartPlug for example) using the built-in popup window with progress bar.

backupartist, 2023-07-24:
Honestly I would have no complaints about a pop up window either. Sometimes function, is easier than form. Function being the key word, as long as it works.. :)

RobinTail, 2023-08-11:
I found that OctoPrint uses this library for tooltips:
https://jqueryui.com/tooltip/#custom-style

RobinTail, 2023-08-13:
The UI will look this way, @backupartist :

[image]

The label will also count the time down.
The popover can be closed by clicking on ✖️ or by clicking outside.
Also, I wanna make those notification and opt-in or opt-out feature.

So, it's going well so far in #171

RobinTail, 2023-08-14:
@backupartist , the feature released in the recently published v3.7.0 🚀

backupartist, 2023-08-16:
I have been gone for a few days with work, and came back to see this update :)   I was excited to try it out, and unfortunately it did not work for me at first, and I am not sure its working properly for me still.   I set up a short print, and when it finished, nothing happened at all.  I rebooted everything and tried again, Still nothing happened. I tried hovering my mouse over the icon and still nothing.  I then tried disabling that option and clicked the save.  then re-enabled it, and clicked save. Then the notification pops up. It only works if you disable it and re-enable it within the timeout period.  I then tried leaving it enabled and then refreshing the browser page, and that worked also. It does not however automatically pop up for me, unless I do those particular things. I an assuming it should automatically pop up without a manual browser refresh.   I have uploaded some pictures of my octorelay version. Also my configuration. I am using Themify with the dark theme, but I don't know if that would cause the problem though.  I have included pictures of all versions.  
[image]
[image]
[image]
[image]

PR #171 (2023-08-14): Feat: Ability to cancel postponed toggling
Closes #87 

### Dialog

[image]

### New setting (opt-out)

[image]

### Tasks

- [x] tests
- [x] countdown: 1 minute / 1 second
- [x] disposer
- [x] opt-in/out these notifications
PR #173 (2023-08-14): 🚀 3.7.0
Closes #87 

- #171