borisbu/OctoRelay #153: Feature: disconnect before turning printer relay OFF

RobinTail, 2023-08-05:
> In a same way autodisconnect would be great. Disconnect printer before relay is deactivated would avoid "SerialException" error from octoprint.

This would require for OctoRelay to know which relay controls the printer.

_Originally posted by @RobinTail in https://github.com/borisbu/OctoRelay/issues/18#issuecomment-1664642386_

Ano10, 2023-08-07:
Could this be simply a check box next to "OFF after printing" like "Disconnect printer before OFF" ?

Ano10, 2023-08-07:
Well I've having only one printer connected to my raspberry pi running octoprint, so I haven't considere case where there would be many printer connected to one raspberry pi. Can it even be possible ? 😅 I though octoprint could only control one printer.

RobinTail, 2023-09-15:
I found a way to disconnect the printer before turning its relay off, @Ano10 .
Works like a charm — no error messages, even in the middle of ongoing printing process.
#207

PR #207 (2023-09-15): Disconnecting from the printer before when turning its relay off
Closes #153 

This feature makes the following improvements:
- it prevents the `SerialException` error appearance when turning the printer relay off,
- it prevents the redundant attempts to connect when there are no printers visible.
PR #201 (2023-09-16): 🚀 3.9.0
Accumulative PR

Closes #153 

- ~~#186~~
- #200 
- #206 
- #203 
- #207
closed 2023-09-16 by RobinTail