borisbu/OctoRelay #154: Relay auto off after X seconds

Tig392, 2023-08-06:
Hi,

Would it be possible to add this function for the printer control relay? My printer is powered on by a momentary action push button. I've extended the wires from this to a relay to power it on remotely but have to click the relay on then off every time. If the relay could auto off after 1 second this would be great.

RobinTail, 2023-08-06:
What kind of printer is this, @Tig392 ?

RobinTail, 2023-08-07:
https://www.creatbot.com/downloads/CreatBot_User_Manual_V7.3_F430_en.pdf

[image]

RobinTail, 2023-08-07:
The problem with sending a short ON-OFF signal to such a device is that it's impossible to figure out its state.
The plugin implementation implies a one-to-one relationship between:

- pin state
- relay state
- device state

Thus, toggling the pin toggles the device and reading the pin state tells the device state.

I'd suppose that there should be a way to connect the relay somewhere deeper to the device circuit instead of its button. There should be some electrical component that finally transforms the pushes of that button into a stable HIGH/LOW voltage.

I'm going to ask my electronics engineer friend about that, @Tig392

RobinTail, 2023-08-07:
Or maybe it would be even easier to make some converter from relay signal to an impulse for that button. I believe I read an article on how to do similar thing to ATX power supply of a PC.

RobinTail, 2023-08-07:
This topic on StackExchange is about similar problem, there are a lot of circuit schemas in comments for different cases.

https://electronics.stackexchange.com/questions/555261/how-to-toggle-relay-with-single-push-button

RobinTail, 2023-08-07:
@Tig392 , could you confirm that there are only two wires connected to the power button?

[image]

Does the manufacturer provide an electrical diagram of the machine?

RobinTail, 2023-08-17:
> > Introducing the new event: After Turned ON.

> Thanks :-)

Does it solve this issue for you, @Tig392

closed 2023-08-21 by RobinTail