borisbu/OctoRelay #187: Sensing

limewebdevelopment, 2023-08-17:
There is another plugin called PSU Control that has a sensing option to monitor the GPIO pins on the Pi.
It gives a real indication of the relay status as opposed to just an On/Off toggle.

Do you think this would be a good feature for OctoRelay too?

[image]

[image]

RobinTail, 2023-08-17:
Yes, I was thinking about that topic due to the following request:
https://github.com/borisbu/OctoRelay/issues/20

However, I don't want to implement and maintain dozens of sensing protocols.
I didn't figure out yet how I could delegate the implementation sensor polling.

I see on your screenshot there is something called "plugin", which is a "plugin for plugin", that could be an approach to that.
But no actual development has been done toward that yet.

Your suggestions and ideas are welcome, @limewebdevelopment

limewebdevelopment, 2023-08-17:
Yeah, this this plugin - https://github.com/kantlivelong/OctoPrint-PSUControl-RPiGPIO

I am not sure if it would be possible to utilise this plugin in it's original form?

RobinTail, 2023-08-17:
I'm sorry — I thought it's about sensors, like temperature/humidity/etc sensors.

This one, however, just reads the pin state.
OctoRelay also doing this. The pin state is reflected on the button in the navigation bar.
The settings `Icon ON` and `Icon OFF` are for reflecting the actual relay state (derived from the pin state + `Inverted Output` setting).

So. Could you please clarify, what exactly you think is missing or should be a good addition? @limewebdevelopment

RobinTail, 2023-08-17:
Perhaps I should clarify that in Readme, that the plugin not only toggles the relays, but also indicates their state.

limewebdevelopment, 2023-08-17:
> Perhaps I should clarify that in Readme, that the plugin not only toggles the relays, but also indicates their state.

That's exactly the bit missing from the readme :) It wasn't clear it did that. Thank you for clarifying.

PR #188 (2023-08-17): Improving documentation: the plugin indicates the relay state too
Closes #187 

This also reflects the recent changes I made in #166
closed 2023-08-17 by RobinTail