borisbu/OctoRelay #189: Is there a setting for this?

backupartist, 2023-08-18:
I am using a relay switched power strip that physically turns everything off.  I love this feature, however it does not work properly for me. Since I use that power strip and its having to completely reboot, it times out before my printer is fully booted up in order to be detected and connected, Is there a delay setting for this, that I can set before it tries to AutoConnect? That would give my printer time to fully boot up.

_Originally posted by @backupartist in https://github.com/borisbu/OctoRelay/issues/18#issuecomment-1679966520_

RobinTail, 2023-08-18:
Hello @backupartist ,

currently there is no setting to postpone AutoConnect. It attempts to connect as soon as it receives the event `ConnectionsAutorefreshed`, which in my case fired when the printer becomes available for connection — I have Anet A6 connected using USB cable to Raspberry Pi.

In issue #18 I also thought that it may turn out that in some cases the delay might be required, but it was not clear at that moment and I decided to release the feature anyway as an MVP. However, now it's clear that the setting to configure the delay is required. Thank you for reporting this problem.

backupartist, 2023-08-19:
Excellent,  that would make it perfect for my situation, and I am sure it will help others out also in other situations as well.  I look forward to seeing this when you get it implemented :)  Thank you.

closed 2023-09-23 by RobinTail

RobinTail, 2023-09-23:
🚀 The feature released in version 3.10.0, @backupartist

backupartist, 2023-09-28:
I just got home from out of town work again, and updated to the latest build, set the setting to what my printer boot up time is, and it works perfectly,  just wonderful.  I couldn't ask for anything more, this plugin works perfectly for what I need.  Thank you again many times over.  Even if I don't need it, I cant wait to see what you add to it next.  :)

PR #217 (2023-09-23): AutoConnect delay
Closes #189 

[image]
PR #218 (2023-09-23): 🚀 3.10.0
Closes #189 

- #217 

[image]