borisbu/OctoRelay #190: Is this operating properly?

backupartist, 2023-08-18:
I have been gone for a few days with work, and came back to see this update :)   I was excited to try it out, and unfortunately it did not work for me at first, and I am not sure its working properly for me still.   I set up a short print, and when it finished, nothing happened at all.  I rebooted everything and tried again, Still nothing happened. I tried hovering my mouse over the icon and still nothing.  I then tried disabling that option and clicked the save.  then re-enabled it, and clicked save. Then the notification pops up. It only works if you disable it and re-enable it within the timeout period.  I then tried leaving it enabled and then refreshing the browser page, and that worked also. It does not however automatically pop up for me, unless I do those particular things. I an assuming it should automatically pop up without a manual browser refresh.   I have uploaded some pictures of my octorelay version. Also my configuration. I am using Themify with the dark theme, but I don't know if that would cause the problem though.  I have included pictures of all versions.  
[image]
[image]
[image]
[image]

_Originally posted by @backupartist in https://github.com/borisbu/OctoRelay/issues/87#issuecomment-1679954064_

backupartist, 2023-08-19:
Yes, I have read all the information, before I posted. My issue is, the popover does not pop up at all, or even in the first place, after the printing stops, I even tried the cancel button, like in your video as a test,  and still no popover.  It is not from clicking OUTSIDE of the popover, I knew that made it disappear, I read the instructions before I tried it. In your video the popover automatically just pops up on the screen, the way I would have assumed it would, Mine however does not.  I have tried it on multiple browsers, cleared the cache, used incognito on chrome. I have rebooted everything, Just short of trying to uninstall and reinstall I don't know why it doesn't work properly for me. I have to refresh my browser webpage manually in order to make it pop up.  I found that works also, instead of just doing the disable and enable method.  Is it possible that it is caused by the themify plugin you think?

RobinTail, 2023-08-19:
> I even tried the cancel button, ..., and still no popover. 

I'd like clarify just for us to be on the same page.
The Cancel button is located inside the popover.

[image]

>  In your video the popover automatically just pops up on the screen... Mine however does not

Okay.
There is also a condition that shows the popover only if the state is going to CHANGE:
meaning that the target state differs from the current state of the relay.

So, if the setting "on Printing Stopped" is set to `OFF` than in order popover to appear that relay should be `ON`.

> Is it possible that it is caused by the themify plugin you think?

Try disabling the Themify plugin and restarting OctoPrint after that.

Anyway. I'm going make a version in a "Prerelease channel", having no listener for the outside click, just to exclude this case to be sure. I will let you know in another message, so we could check it together.

Thank you, @backupartist

RobinTail, 2023-08-19:
Meanwhile, I'm going to try installing Themeify myself with the dark theme you have.
Maybe I'll find out something.

RobinTail, 2023-08-19:
No, with the default settings of Themeify the popover appears in my case. It looks ugly, but it appears :)

[image]

RobinTail, 2023-08-19:
I also tried it for the printer's relay, for the same event with the same settings from your screenshot.
So, it's not the event or relay specific issue.

[image]

[image]

RobinTail, 2023-08-19:
Dear @backupartist ,

let's try the following in order to check or exclude the hypothesis of outside click.

- Please proceed to the Software update section of the OctoPrint settings.
- Find and OctoRelay plugin there and set the "Prerelease" option for the "Release channel" setting

[image]

- On the top of the window click on "Force check for updated" (in the dropdown button).
- Please try to install the version `3.8.1rc1` that has no outside click listener.
- Repeat the steps from you case.

Then please let me know if the popover did appear or not.
In this version the popover can only be dismissed by clicking on `x` button within it.

RobinTail, 2023-08-21:
@backupartist , still waiting for you to check the version 3.8.1rc1

backupartist, 2023-08-22:
Sorry I was away for work again..  I just tried all of that and still does not pop up, unless I refresh the browser, I tried with Edge, and Chrome, including incognito, just to make sure it was not a hung cache. Still the same thing. The popup does not pop up unless I manually refresh the web browser page.  I have no clue, everything else works just fine.  I have included pictures of my octorelay version, and my themify disabled.  you can see the standard white background, Trying to keep the install as default as possible. Could it be an inherent problem with my octoprint version?
[image]
[image]

backupartist, 2023-08-22:
I should also clarify, The popup does pop up,  but only after i refresh the web browser page. Hitting F5 after the print stops, makes it pop up, but it does not do it automatically after the print stops.

backupartist, 2023-08-22:
As another test, I just made the fan relay active, with the same settings as you have on your picture from 3 days ago, for your test. The one you confirmed that did work. I still got the same thing. Thee popup for the printer and the fan neither one automatically popped up. Once I hit the F5 key, and the webpage refreshed, both of them popped up. I have no idea now.. lol

RobinTail, 2023-08-22:
Okay. Thank you for checking that, @backupartist 

> Could it be an inherent problem with my octoprint version?

By the way, what is your OctoPrint version?

RobinTail, 2023-08-22:
> I tried with Edge, and Chrome, including incognito
> Once I hit the F5 key, and the webpage refreshed, both of them popped up.

Let's try checking a possible JS issue:

1. Before running your test case in Chrome, please open the Developer Console:
  - Assuming that you're on Windows, it should be by pressing `Ctrl + Shift + J`
  - https://balsamiq.com/support/faqs/browserconsole/#google-chrome
  - https://developer.chrome.com/docs/devtools/open/

It should look this way, please ensure the tab "Console" is activated:
[image]

2. Then please retry the steps that should lead to the appearance of the popover.
3. If something appears in the console instead of showing the popover, please copy/paste that to this thread.

RobinTail, 2023-08-22:
One more important question, @backupartist : 

- When you expect the popover to appear, but it does not, do you see the tooltip (black hint) when hover the pointer over the button?

backupartist, 2023-08-25:
This is my version...
[image]

I tested again in Chrome, in incognito, and regular, after I updated to your latest update. Still did not automatically popup without F5 refresh.  I opened the console and ran multiple tests, there were no errors pertaining to your plugin.  I got two errors, one from octroeverywhere, and one from prettygcodeviewer. I disabled both of those plugins and tried everything again, and still nothing, still have to refresh. I disabled a few more plugins just to see if anything would change, and it is still doing the same thing.

- When you expect the popover to appear, but it does not, do you see the tooltip (black hint) when hover the pointer over the button?

Yes I do,     The print stops, The cancel popup does not pop up. I mouse over the icon for the printer relay, and the black tooltip pops up and says "Printer".  I then hit F5, and the cancel popup pops up, then I can cancel.  I can also X out of the cancel popup after F5, and make it disappear. Then hit F5 again, do a refresh, and the cancel popup pops back up.  I mean it does work, I just have to watch the printer physically or on the cam, or the print time to see when it stops,  Then hit F5 to refresh the page to make it pop up instead of automatically.  At least that is better than having to wait for it to turn off, to turn it on again.  I just wish I knew what was causing it. Mildly irritating, but more so a mystery that makes me wonder what is causing it. 

  I am gong to send a snapshot of all my plugins. I dont know if there may be some issue with one of them making the popup not work. You can see what I have installed.

backupartist, 2023-08-25:
[image]
[image]
[image]
[image]

RobinTail, 2023-08-25:
Thank you, @backupartist .

I'd like to investigate this issue further.

1. Please proceed to the Logging section of the OctoPrint settings.
2. Delete the logs file `octoprint.log`
3. In the Logging Levels section find the `octoprint.plugins.octorelay` option and choose the level `DEBUG`.
4. Click Save
5. Restart OctoPrint (that will write your plugin settings in the log)
6. Repeat your test (the actions that should lead to showing the popover on upcoming switch)
7. Proceed to Logging again
8. Download the file `octoprint.log` and share it here
   1. <details><summary>or share it privately</summary>robin_tail@protonmail.com</details>

[image]

RobinTail, 2023-08-26:
Dear @backupartist ,

thank you very much for sending me your logs.
Good news — I think I finally understand why it's happening to you.
In short: the UI update is missing — this is a bug.

### Long story.

This is the moment when you finished printing:

```
Received the PrintDone event having payload ***
Handling the plugin event PRINTING_STOPPED having scope: dict_keys(['r1', 'r2', 'r3', 'r4', 'r5', 'r6', 'r7', 'r8'])
Cancelling tasks by request from PRINTING_STOPPED for relay r2
The cancelled tasks removed from the registry
Postponing the switching of the relay r2 by 300s
The task registered: Task(r2,False,PRINTING_STOPPED,300)
```

It registers the task, but then... nothing happened.
So then next log entry is when you refreshed the page

```
Received the UserLoggedIn event having payload: ***
```

Which triggers

```
Received the ClientOpened event having payload: ***
Updating the UI
Finding the upcoming tasks
The UI feed: {
    'r1': ***, 
    'r2': { ***, 'upcoming': {'target': False, 'owner': 'PRINTING_STOPPED', 'deadline': 1693034330972} }, 
    'r3': ***,
    ***
}
```

So the UI is not being updated during the handling of the PRINTING_STOPPED event.
I was wondering why it works in my case and the reason is that I have another relay that changes the state by the same event. This switch happens at the same time and it triggers the UI update for me. But in your case there are no relay toggling in the event of PRINTING_STOPPED. Therefore this issue happens.

I'm going to fix it ASAP.
Thank you once again for the patience and elaboration.

RobinTail, 2023-08-26:
🚀  Fixed in 3.8.2rc1, @backupartist 
The stable release is coming soon today as well.

closed 2023-08-26 by RobinTail

backupartist, 2023-09-01:
So I just got home again from being out of town for work.  I got some time to do some printing. I downloaded the new latest octorelay update, I set up my disc test print. I let it finish printing, which didn't take long, and then lo and behold....  This automatically popped up :)
[image]
It is amazingly working perfectly now. Woot woot..   
I have to say, thank you for all the hard work you are doing for this plugin. Other than the Dashboard plugin, Octorelay is my most used plugin for all my remote controlled operations to my printer. Thanks to your hard work, this plugin is becoming the perfect tool for remote operations. Again I say, Thank you Thank you.

RobinTail, 2023-09-01:
I'm glad to read it, @backupartist . Thank you for the feedback.

PR #192 (2023-08-19): Removing the outside click listener for upcoming switch popover.
Probably fixes #190 

This is the UX adjustment that removes the outside click listener from the popover, that might lead to confusion, making the popover disappear unexpectedly from the user's perspecive.
PR #196 (2023-08-26): Add missing UI update
Fixes #190
PR #197 (2023-08-26): 🚀 3.8.2
Fixes #190 

- #196