borisbu/OctoRelay #195: Unable to update

Targa2405, 2023-08-24:
I'm unable to update from version 2.2.2 to the latest version, each time I get the same error-message;

"WARNING: Built wheel for OctoRelay is invalid: Metadata 1.2 mandates PEP 440 version, but 'unknown-g1f04e4f3' is not
ERROR: Could not build wheels for OctoRelay, which is required to install pyproject.toml-based projects"

RobinTail, 2023-08-24:
Sorry for the inconvenience, @Targa2405 .

1. Please proceed to the OctoRelay settings
2. Go to Plugin Manager
3. Click on Get More button
4. Scroll the window down to `... from URL` section
5. Copy/paste the following URL:

```
https://github.com/borisbu/OctoRelay/releases/latest/download/release.zip
```

6. And click on the Install button next to it.

This should solve the issue for you.

[image]

ak185158, 2023-08-26:
This worked to resolve the error for me as I was having the same issue with this release.

RobinTail, 2023-08-26:
Thank you for letting me know, @ak185158 .
There will be another bugfix today — #197 
I hope this time you'll be able to update automatically.
If not, please open another issue.

closed 2023-08-26 by RobinTail