borisbu/OctoRelay #204: Feature/Help: Change Icons

yani66, 2023-09-03:
Hi,
I wanted to change the Icons shown in the navbar, but I couldn't find the folder of the Icon-files on the raspberry.
It would be great to know where the icons are saved.
The feature to select from a greater variety of icons and/or upload own icons, would be a great addition.

Thanks in advance and for the plugin so far
yani66

RobinTail, 2023-09-03:
Hello @yani66 ,

> where the icons are saved

You can place your icons into the plugin's `static/img` folder under the OctoPrint's virtual environment.
In my particular case it is:

```sh
~/oprint/lib/python3.9/site-packages/octoprint_octorelay/static/img
```

Then you can specify the icon HTML tag this way:

```html
<img width="24" height="24" src="/plugin/octorelay/static/img/3d-printer.svg">
```

> The feature to select from a greater variety of icons and/or upload own icons, would be a great addition.

The desired variety of icons can not be included into the distribution since most of them require to comply a license.
Moreover, it would unreasonably increase the distribution weight.

Nevertheless, already now you can select from the variety of icons from the Internet.
Type `printer icon svg` in Google and copy the URL of the one you like.
You can place that URL into the `src` property of the `img` tag above instead of copying that file onto Raspberry Pi.

I hope this information will help you

RobinTail, 2024-02-18:
Possible solution is via allowing to upload anything
https://github.com/rlogiacco/UploadAnything/blob/master/octoprint_uploadanything/__init__.py

SCH00N3R, 2024-11-05:
It isn't strictly necessary to upload the image to octoprint. You could host it on an internal web server or link to another website that hosts the image you wish to use. Of course under that scenario you are at the mercy of the external site as to whether the site will exist years from now or if the image path will change. Nonetheless I thought I would mention it. @RobinTail 's suggestion is better.

Hydrop75, 2025-02-22:
On 4.x version i'm connect to the PI
(with Cyberduckfor mac  or winscp on windows)
Then browse
/home/pi/oprint/lib/python3.7/site-packages/octoprint_octorelay/static/img
then send to this directory

[image]
[image]

and to the plugin change the names of target files
I hope it could help