borisbu/OctoRelay #205: Auto-off timer is not canceled with a new print

CrNMGuy, 2023-09-08:
When send a new print following a previous print, autooff timer is not canceled. Noticed through the new tootip "printer goes OFF in xx minutes [cancel]" while printing. When print end, autooff is reset.
Expected beahaviour : while printer is on an connected, if i send a new print, it should cancel autotimer, wait for print to end then init timer for autooff

RobinTail, 2023-09-08:
@CrNMGuy ,

currently when start printing all previously postponed switches are cancelled, there are no exceptions.
Though, the new one can be set by the same event.

Since it's not clear what exactly and why is happening, please provide the debug level logs of the plugin (enable it in Logging section of OctoPrint) and a screenshot of your printer relay configuration.

RobinTail, 2023-09-18:
Managed to reproduced on v3.9.0, @CrNMGuy .

```
2023-09-19 00:24:16,344 - octoprint.plugins.octorelay - DEBUG - Handling the plugin event PRINTING_STARTED having scope: ['r1', 'r2', 'r3', 'r4', 'r5', 'r6', 'r7', 'r8']
2023-09-19 00:24:16,573 - octoprint.plugins.octorelay - DEBUG - Cancelling tasks by request from PRINTING_STARTED for relay r1
2023-09-19 00:24:16,641 - octoprint.plugins.octorelay - DEBUG - The cancelled tasks removed from the registry
```

Probably the issue is happening because of the event setting `on Printing Started` and its default value `skip`.

[image]

RobinTail, 2023-09-18:
Yeah, when the `target` is `None`, the postponed tasks are not cancelled.

https://github.com/borisbu/OctoRelay/blob/15818745ade5dc2fb6adf0c101394097ef461619/octoprint_octorelay/__init__.py#L191-L196

CrNMGuy, 2023-09-19:
Sorry, didn't sent debug yet.
But you find the thing :)
Thanks !

closed 2023-09-20 by RobinTail

RobinTail, 2023-09-20:
🚀  The issue fixed in v3.9.1, @CrNMGuy 
Thank you for the contrubution!

PR #213 (2023-09-20): Fix task cancellation when `on Printing Started` is `skip`
Fixes #205
PR #214 (2023-09-20): Ensure UI update when event just cancels tasks
Closes #205 
Addition to #213 
Revealed during the QA in #209
PR #209 (2023-09-20): 🚀 3.9.1
Accumulating PR

Fixes #205 

- #208
- #213 
- #214