borisbu/OctoRelay #210: Suggestions : Power ON on API upload&print

CrNMGuy, 2023-09-19:
PSU Control provide a useful feature that is powering up the printer on a API event which is "Print"
Would like this feature :)

RobinTail, 2023-09-20:
- There is no event "Print". Here is the list of OctoPrint events: https://docs.octoprint.org/en/master/events/index.html#available-events
- If it would exist, how the event "Print" could happen on unpowered printer, @CrNMGuy ?

CrNMGuy, 2023-09-20:
Here is how PSU Control check for upload (don't knowing python & octoprint context)
https://github.com/kantlivelong/OctoPrint-PSUControl/blob/c0490c8c30388a649cc6e87525241dfe3b5923ca/octoprint_psucontrol/__init__.py#L586

             flask.request.path.startswith('/api/files/') and
             flask.request.method == 'POST' and
             flask.request.values.get('print', 'false') in valid_boolean_trues):

About Power ON on print event : 
Still mentionning PSU Control as it upgraded a lot my print experience : When powering ON, plugin seems to activate a Connect to printer and wait while not connected.

RobinTail, 2023-09-20:
Ok. Now I understand.
This is `Upload` event having `print` option in its payload, @CrNMGuy 

[image]

ekammerloher, 2026-09-02:
I have the same need as @CrNMGuy, but did not want to use PSU Control, which has not been updated for a while.

I created a minimal OctoRelay fork (https://github.com/ekammerloher/OctoRelay/tree/helper-api) that only exposes relay helpers for other plugins.
A small companion plugin (https://github.com/ekammerloher/OctoPrint-OctoRelayConnectAndPrint) handles upload-and-print requests, powers the relay, waits for the printer, connects OctoPrint, and starts the print. This keeps concerns separate.

If desired, I can open a PR containing only the helper API, with no other changes.