borisbu/OctoRelay #211: Suggestion : Relay ON on Print, with a timeout

CrNMGuy, 2023-09-19:
My own use-case : when printing at night, i check first layers of my print before going to bed.
Having Light ON for the first 3 layers or first 5 minutes would be useful (then light off)

This might complicate the brand new layout

RobinTail, 2023-09-19:
This should be already achievable using setting `after Turned ON`.
For your light relay you can set to `OFF` with a `300` seconds delay.
If you also set `on Printing Started` to `ON` then your lamp will shine for the first 5 minutes after you started printing, @CrNMGuy

RobinTail, 2023-09-19:
> how to help?

If it's easier for you to do than explain:

1. Fork the repo.
2. In your repo create a branch for the feature/fix/change (ideally starting branching it from `develop`).
3. Make changes to a file (`README.md` for documentation, `templates/octorelay_settings.jinja2` for settings, etc)
4. Commit and push those changes to the branch in your fork
5. Create a PR across forks: from the feature branch of your fork to `develop` branch of this repo.

Otherwise:

- Tell me what you believe should be changed in an/the issue,

@CrNMGuy

closed 2023-09-21 by CrNMGuy