borisbu/OctoRelay #222: cant install

lechiel, 2023-09-27:
Hi, 

Im trying to install the plugin but afther restart their is nothing installed. 

logs: ```
Installing plugin "OctoRelay" from https://github.com/borisbu/OctoRelay/releases/latest/download/release.zip...
/home/pi/oprint/bin/python -m pip --disable-pip-version-check install file:///tmp/tmpoh_3epq2/release.zip --no-cache-dir
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Processing /tmp/tmpoh_3epq2/release.zip
Preparing metadata (setup.py): started
Preparing metadata (setup.py): finished with status 'done'
Requirement already satisfied: OctoPrint in /home/pi/oprint/lib/python3.9/site-packages (from OctoRelay==3.10.0) (1.9.2)
Requirement already satisfied: RPi.GPIO in /home/pi/oprint/lib/python3.9/site-packages (from OctoRelay==3.10.0) (0.7.1)
Requirement already satisfied: wheel in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.34.2)
Requirement already satisfied: future<1,>=0.18.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.18.3)
Requirement already satisfied: cachelib<0.11,>=0.10.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.10.2)
Requirement already satisfied: PyYAML<7,>=6.0.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (6.0.1)
Requirement already satisfied: Babel<2.13,>=2.12.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.12.1)
Requirement already satisfied: Flask-Assets<3,>=2.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.0)
Requirement already satisfied: regex in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2022.9.13)
Requirement already satisfied: passlib<2,>=1.7.4 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.7.4)
Requirement already satisfied: Flask-Login<0.7,>=0.6.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.6.2)
Requirement already satisfied: flask<2.3,>=2.2.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.2.5)
Requirement already satisfied: OctoPrint-FileCheck>=2021.2.23 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2021.2.23)
Requirement already satisfied: requests<3,>=2.28.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.31.0)
Requirement already satisfied: blinker<2,>=1.6.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.6.2)
Requirement already satisfied: OctoPrint-PiSupport>=2023.5.24 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2023.7.31)
Requirement already satisfied: filetype<2,>=1.2.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.2.0)
Requirement already satisfied: psutil<6,>=5.9.4 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (5.9.5)
Requirement already satisfied: pydantic<2,>=1.10.7 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.10.11)
Requirement already satisfied: netifaces<1,>=0.11 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.11.0)
Requirement already satisfied: semantic-version<3,>=2.10.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.10.0)
Requirement already satisfied: markdown<4,>=3.4.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (3.4.3)
Requirement already satisfied: sentry-sdk<2,>=1.19.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.28.1)
Requirement already satisfied: watchdog<3,>=2.3.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.3.1)
Requirement already satisfied: zeroconf==0.39.4 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.39.4)
Requirement already satisfied: emoji<3,>=2.2.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.6.0)
Requirement already satisfied: wrapt<1.16,>=1.15 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.15.0)
Requirement already satisfied: feedparser<7,>=6.0.10 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (6.0.10)
Requirement already satisfied: pyserial<4,>=3.5 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (3.5)
Requirement already satisfied: netaddr<0.9,>=0.8 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.8.0)
Requirement already satisfied: pylru<2,>=1.2.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.2.1)
Requirement already satisfied: werkzeug<2.3,>=2.2.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.2.3)
Requirement already satisfied: websocket-client<2,>=1.5.1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.6.1)
Requirement already satisfied: argon2-cffi<22,>=21.3.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (21.3.0)
Requirement already satisfied: Click<9,>=8.1.3 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (8.1.3)
Requirement already satisfied: sarge==0.1.7.post1 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.1.7.post1)
Requirement already satisfied: Flask-Limiter<4,>=3.3.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (3.3.1)
Requirement already satisfied: zipstream-ng<2.0.0,>=1.5.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.6.0)
Requirement already satisfied: unidecode in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.3.6)
Requirement already satisfied: pkginfo<2,>=1.9.6 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (1.9.6)
Requirement already satisfied: class-doc<0.3,>=0.2.6 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (0.2.6)
Requirement already satisfied: Flask-Babel<4,>=3.1.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (3.1.0)
Requirement already satisfied: frozendict<3,>=2.3.7 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.3.8)
Requirement already satisfied: OctoPrint-FirmwareCheck>=2021.10.11 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2021.10.11)
Requirement already satisfied: tornado<7,>=6.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (6.2)
Requirement already satisfied: pathvalidate<3,>=2.5.2 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (2.5.2)
Requirement already satisfied: colorlog<7,>=6.7.0 in /home/pi/oprint/lib/python3.9/site-packages (from OctoPrint->OctoRelay==3.10.0) (6.7.0)
Requirement already satisfied: ifaddr>=0.1.7 in /home/pi/oprint/lib/python3.9/site-packages (from zeroconf==0.39.4->OctoPrint->OctoRelay==3.10.0) (0.2.0)
Requirement already satisfied: async-timeout>=4.0.1 in /home/pi/oprint/lib/python3.9/site-packages (from zeroconf==0.39.4->OctoPrint->OctoRelay==3.10.0) (4.0.2)
Requirement already satisfied: argon2-cffi-bindings in /home/pi/oprint/lib/python3.9/site-packages (from argon2-cffi<22,>=21.3.0->OctoPrint->OctoRelay==3.10.0) (21.2.0)
Requirement already satisfied: more-itertools>=5.0.0 in /home/pi/oprint/lib/python3.9/site-packages (from class-doc<0.3,>=0.2.6->OctoPrint->OctoRelay==3.10.0) (9.1.0)
Requirement already satisfied: sgmllib3k in /home/pi/oprint/lib/python3.9/site-packages (from feedparser<7,>=6.0.10->OctoPrint->OctoRelay==3.10.0) (1.0.0)
Requirement already satisfied: Jinja2>=3.0 in /home/pi/oprint/lib/python3.9/site-packages (from flask<2.3,>=2.2.3->OctoPrint->OctoRelay==3.10.0) (3.1.2)
Requirement already satisfied: importlib-metadata>=3.6.0 in /home/pi/oprint/lib/python3.9/site-packages (from flask<2.3,>=2.2.3->OctoPrint->OctoRelay==3.10.0) (5.0.0)
Requirement already satisfied: itsdangerous>=2.0 in /home/pi/oprint/lib/python3.9/site-packages (from flask<2.3,>=2.2.3->OctoPrint->OctoRelay==3.10.0) (2.1.2)
[... cut ...]
Created wheel for OctoRelay: filename=OctoRelay-3.10.0-py2.py3-none-any.whl size=35445 sha256=10b90044fc7d0a05b8a8b2b5254f41207aab163a9d7c8c4dd06c018e2a744042
Stored in directory: /tmp/pip-ephem-wheel-cache-qrxedaja/wheels/ce/3c/3b/f48178330090156a427fe397322392e67954e07bee7b2d2ce2
Successfully built OctoRelay
Installing collected packages: OctoRelay
Attempting uninstall: OctoRelay
Found existing installation: OctoRelay 3.10.0
Uninstalling OctoRelay-3.10.0:
Successfully uninstalled OctoRelay-3.10.0
Successfully installed OctoRelay-3.10.0
Done!
```

RobinTail, 2023-09-27:
This is interesting, @lechiel 

```
Successfully built OctoRelay
Installing collected packages: OctoRelay
Attempting uninstall: OctoRelay
```

So, after building it's trying to install, but then immediately uninstalling it.
I'd assume that there may be some writing permissions issue.

Normally it should look this way (in your case RPi.GPIO is already installed):

```
Successfully built OctoRelay
Installing collected packages: OctoRelay
Successfully installed OctoRelay-3.10.0
```

I also noticed the following entry in your log:
```
Stored in directory: /tmp/...
```
. which does not exist in mine.

Consider trying the following approach:
- Download the latest release file from https://github.com/borisbu/OctoRelay/releases/latest/download/release.zip
- Use manual installation "from uploaded file" instead

[image]

Please let me know if this worked for you or not.
Consider checking OctoPrint forums for similar issues. Maybe it's still a well-known writing permissions issue.

RobinTail, 2023-09-27:
Found similar issue: https://www.reddit.com/r/octoprint/comments/13t0b5x/issue_installing_astroprint_plugin/

RobinTail, 2023-09-27:
@lechiel , please also check what the UI says.
If it says "Successfully Installed: Unknown", proceed to this issue:
https://community.octoprint.org/t/plugin-installation-unknown/1636

RobinTail, 2023-09-27:
Another similar issue (seems to be unresolved): https://community.octoprint.org/t/octolapse-installation-help/25992

RobinTail, 2023-09-27:
@lechiel , please also check this out: https://community.octoprint.org/t/astroprint-plugin-install-issue/52026/10?u=bcmmakers

Most likely it can be a Python environment issue

RobinTail, 2023-09-27:
@lechiel ,

I also noticed that "uninstall" happens when updating the plugin.
Perhaps there is some duplicated installation.
I think it may worth it to check if you have the following directory (or similar):
`~/oprint/lib/python3.9/site-packages/octoprint_octorelay`

[image]

lechiel, 2023-09-28:
Thanks for all your comments. Its printing now so when it finished i will try all the suggestions

RobinTail, 2023-09-30:
@lechiel , did anything help or is there any update on this issue?

lechiel, 2023-09-30:
I tried different things. Didn't work so I did a reinstall. Works now

RobinTail, 2023-09-30:
Great. What was your approach to reinstall?
Did you use the downloaded file?
This may help people having similar issue, @lechiel

lechiel, 2023-09-30:
I did a clean instal of octoprint.

closed 2023-09-30 by RobinTail