borisbu/OctoRelay #230: Can’t Update or Uninstall. Stuck on 3.9.1.

davidarmenphoto, 2023-09-28:
Currently using 3.9.1. I tried updating to 3.10 a few days ago and it would say ”Done” but the version would stay 3.9.1 in the plug-in manager after the restart. I didn’t give it much thought. But today, I got the notification about 3.11.1 and I clicked update. Same issue, it says done at the end but the plug-in is still on 3.9.1. 

Then I tried to completely uninstall the plugin so I can do a fresh install. It will not even Uninstall. I get some error so I can’t even uninstall the app now. What do I do?

[image]
[image]

RobinTail, 2023-09-29:
@davidarmenphoto ,

on your first screenshot I noticed that the last parameter supplied to `pip` is the plugin name without its first letter: `-ctoRelay`, or having first letter replaced with a dash.

RobinTail, 2023-09-29:
The corresponding line of code in OctoPrint repo:
https://github.com/OctoPrint/OctoPrint/blob/ed4a2646fb4e2904892c895580192987242834c8/src/octoprint/cli/dev.py#L246

what is your version of OctoPrint, @davidarmenphoto ?

RobinTail, 2023-09-29:
Found similar case on OctoPrint forum, @davidarmenphoto : 

https://community.octoprint.org/t/plugin-uninstall-has-corrupted-name-for-commandline-plugin-problem-or-octoprint-problem/29446

RobinTail, 2023-09-29:
I can confirm that there is no such typo as `-ctoRelay` in the distributable package of version 3.9.1.

[image]

RobinTail, 2023-09-29:
The first letter being replaced by `-` is most likely an issue of the following python module:

https://github.com/pallets/click

RobinTail, 2023-09-30:
@davidarmenphoto , consider checking version of the `click`:

```bash
> pip list | grep click
click                     8.1.6
```

if it's too old (lastest is 8.1.7), consider upgrading it:

```bash
> pip install click -U
Requirement already satisfied: click in /opt/python/3.10.8/lib/python3.10/site-packages (8.1.6)
Collecting click
  Downloading click-8.1.7-py3-none-any.whl (97 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 97.9/97.9 kB 2.4 MB/s eta 0:00:00
Installing collected packages: click
  Attempting uninstall: click
    Found existing installation: click 8.1.6
    Uninstalling click-8.1.6:
      Successfully uninstalled click-8.1.6
Successfully installed click-8.1.7
```

in your case instead of `pip` you should write `/home/pi/oprint/bin/python -m pip`

jneilliii, 2023-10-01:
I'd recommend trying to determine if it's a messed up python lib scenario. What is the result of the command 
```
/home/pi/oprint/bin/pip freeze | grep Relay
```
For whatever one is returned do the command
```
/home/pi/oprint/bin/pip uninstall <put name returned from above command>
```

davidarmenphoto, 2023-10-01:
I’m incredibly grateful for the extremely detailed responses! My apologies but unfortunately, I backed up and wrote a new octopi image. The restore downloaded the latest version of OctoRelay from the repo and its been flawless since.

Also, even after a fresh octopi install, I checked the version of `click` and it was not current, it was much older than 8.1.7. I cant remember but it might have been 8.0.9. I upgraded it to the latest version just to prevent and future issues.

closed 2023-10-02 by RobinTail