borisbu/OctoRelay #232: Icon on UI taskbar not displayed after Upload&Print API command

CrNMGuy, 2023-09-30:
How to reproduce : 
(no print) opening octopi webpage > ui icon are diaplayed
closing octopi webpag
(api command upload and print) opening octopi webpage > NO ui icon 

How to fix ? restart octopi

RobinTail, 2023-09-30:
I was not able to reproduce this case, @CrNMGuy 
I sent a file to print from Cura with an option "start printing", then opened OctoPrint UI and all my configured icons in the navigation bar were visible and operational.

Consider following:
1. Before opening OctoPrint UI, open the browser console in the same window — check for errors when you open the UI.
2. Download the OctoPrint log and find the portion of the messages logged from the moment when you sent the API command. Consider switching `debug` mode for the OctoRelay logging. There should be records on what was the payload sent by the plugin to its UI.
3. Share your findings.

CrNMGuy, 2023-10-01:
Nothing in octoprint.log with debug activated

When Inspecting webpage, seems like UI JS script is not launched
[image]

RobinTail, 2023-10-02:
> BUT when waiting ~60-100sec, icons showed up

It's interesting, because it means that some other task is delaying the execution of the plugin's code.

I'd still like to ask you for the OctoPrint log entries before and including the line having text "The UI feed:". The debug level for the plugin is required for that.
This entry is placed to the log right before sending the information on your buttons to the UI.
By inspecting the entries logged before it, we could probably figure out what causes the delay. @CrNMGuy

CrNMGuy, 2023-10-02:
Sure, here it is : 
```
2023-10-02 19:27:14,797 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-10-02 19:37:00,583 - octoprint.access.users - INFO - Cleaning up user session 5DD250A9AECD4693A287D34E57342E25 for user florian
2023-10-02 19:37:00,584 - octoprint.access.users - INFO - Logged out user: florian
2023-10-02 19:37:00,633 - octoprint.server.util.flask - INFO - Passively logging in user florian from ::ffff:192.168.102.71
2023-10-02 19:37:00,634 - octoprint.access.users - INFO - Logged in user: florian
2023-10-02 19:37:00,641 - octoprint.plugins.octorelay - DEBUG - Received the UserLoggedIn event having payload: {'username': 'florian'}
2023-10-02 19:37:00,749 - octoprint.server.util.flask - INFO - Passively logging in user florian from ::ffff:192.168.102.71
2023-10-02 19:37:00,750 - octoprint.access.users - INFO - Logged in user: florian
2023-10-02 19:37:00,756 - octoprint.plugins.octorelay - DEBUG - Received the UserLoggedIn event having payload: {'username': 'florian'}
2023-10-02 19:37:00,848 - octoprint.server.util.flask - INFO - Passively logging in user florian from ::ffff:192.168.102.71
2023-10-02 19:37:00,849 - octoprint.access.users - INFO - Logged in user: florian
2023-10-02 19:37:00,855 - octoprint.plugins.octorelay - DEBUG - Received the UserLoggedIn event having payload: {'username': 'florian'}
2023-10-02 19:37:01,662 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.102.71
2023-10-02 19:37:01,707 - octoprint.plugins.octorelay - DEBUG - Received the ClientOpened event having payload: {'remoteAddress': '::ffff:192.168.102.71'}
2023-10-02 19:37:01,707 - octoprint.plugins.octorelay - DEBUG - Updating the UI
2023-10-02 19:37:01,808 - octoprint.server.util.flask - INFO - Passively logging in user florian from ::ffff:192.168.102.71
2023-10-02 19:37:01,808 - octoprint.access.users - INFO - Logged in user: florian
2023-10-02 19:37:01,815 - octoprint.plugins.octorelay - DEBUG - Finding the upcoming tasks
2023-10-02 19:37:01,818 - octoprint.plugins.octorelay - DEBUG - The UI feed: {'r1': {'relay_pin': 14, 'inverted_output': True, 'relay_state': False, 'label_text': 'Light', 'active': True, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r2': {'relay_pin': 15, 'inverted_output': True, 'relay_state': False, 'label_text': 'Printer', 'active': True, 'icon_html': '<img width="24" height="24" src="/plugin/octorelay/static/img/3d-printer.svg" style="filter: opacity(20%)">', 'confirm_off': False, 'upcoming': None}, 'r3': {'relay_pin': 18, 'inverted_output': True, 'relay_state': False, 'label_text': 'Fan', 'active': False, 'icon_html': '<img width="24" height="24" src="/plugin/octorelay/static/img/fan.svg" style="filter: opacity(20%)">', 'confirm_off': False, 'upcoming': None}, 'r4': {'relay_pin': 23, 'inverted_output': True, 'relay_state': False, 'label_text': 'Webcam', 'active': False, 'icon_html': '<img width="24" height="24" src="/plugin/octorelay/static/img/webcam.svg" style="filter: opacity(20%)">', 'confirm_off': False, 'upcoming': None}, 'r5': {'relay_pin': 24, 'inverted_output': True, 'relay_state': False, 'label_text': 'R5', 'active': False, 'icon_html': 'OFF', 'confirm_off': False, 'upcoming': None}, 'r6': {'relay_pin': 25, 'inverted_output': True, 'relay_state': False, 'label_text': 'R6', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r7': {'relay_pin': 8, 'inverted_output': True, 'relay_state': False, 'label_text': 'R7', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r8': {'relay_pin': 7, 'inverted_output': True, 'relay_state': False, 'label_text': 'R8', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}}
2023-10-02 19:37:01,876 - octoprint.plugins.octorelay - DEBUG - Received the UserLoggedIn event having payload: {'username': 'florian'}
2023-10-02 19:37:05,255 - octoprint.server.util.sockjs - INFO - User florian logged in on the socket from client ::ffff:192.168.102.71
2023-10-02 19:37:05,266 - octoprint.plugins.octorelay - DEBUG - Received the ClientAuthed event having payload: {'username': 'florian', 'remoteAddress': '::ffff:192.168.102.71'}
2023-10-02 19:37:06,560 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:37:06,659 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:37:06,760 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:37:06,879 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:37:06,980 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:37:07,728 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from https://plugins.octoprint.org/plugins.json
2023-10-02 19:37:45,426 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.102.71
2023-10-02 19:37:45,431 - octoprint.plugins.octorelay - DEBUG - Received the ClientClosed event having payload: {'remoteAddress': '::ffff:192.168.102.71'}
[... cut ...]
2023-10-02 19:40:11,006 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:40:11,080 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: None
2023-10-02 19:41:16,424 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.102.71
2023-10-02 19:41:16,429 - octoprint.plugins.octorelay - DEBUG - Received the ClientClosed event having payload: {'remoteAddress': '::ffff:192.168.102.71'}
```

CrNMGuy, 2023-10-02:
This timing seems the delay for update UI : 
`2023-10-02 19:39:29,153 - octoprint.filemanager.analysis - INFO - Analysis of entry local:3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode finished, needed 82.81s
`

CrNMGuy, 2023-10-02:
This filemanager seems to delay OctoRelay UI update

```
2023-10-02 19:47:13,425 - octoprint.filemanager.analysis - INFO - Starting analysis of local:3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode
2023-10-02 19:48:38,225 - octoprint.filemanager.analysis - INFO - Analysis of entry local:3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode finished, needed 84.80s
2023-10-02 19:49:18,243 - octoprint.plugins.octorelay - DEBUG - The UI feed: {'r1': {'relay_pin': 14, 'inverted_output': True, 'relay_state': False, 'label_text': 'Light', 'active': True, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r2': {'relay_pin': 15, 'inverted_output': True, 'relay_state': False, 'label_text': 'Printer', 'active': True, 'icon_html': '<img width="24" height="24" src="/plugin/octorelay/static/img/3d-printer.svg" style="filter: opacity(20%)">', 'confirm_off': False, 'upcoming': None}, 'r3': {'relay_pin': 18, 'inverted_output': True, 'relay_state': False, 'label_text': 'Fan', 'active': False, 'icon_html': '<img width="24" height="24" src="/plugin/octorelay/static/img/fan.svg" style="filter: opacity(20%)">', 'confirm_off': False, 'upcoming': None}, 'r4': {'relay_pin': 23, 'inverted_output': True, 'relay_state': False, 'label_text': 'Webcam', 'active': False, 'icon_html': '<img width="24" height="24" src="/plugin/octorelay/static/img/webcam.svg" style="filter: opacity(20%)">', 'confirm_off': False, 'upcoming': None}, 'r5': {'relay_pin': 24, 'inverted_output': True, 'relay_state': False, 'label_text': 'R5', 'active': False, 'icon_html': 'OFF', 'confirm_off': False, 'upcoming': None}, 'r6': {'relay_pin': 25, 'inverted_output': True, 'relay_state': False, 'label_text': 'R6', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r7': {'relay_pin': 8, 'inverted_output': True, 'relay_state': False, 'label_text': 'R7', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r8': {'relay_pin': 7, 'inverted_output': True, 'relay_state': False, 'label_text': 'R8', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}}

```

```
2023-10-02 19:42:14,799 - octoprint.server.heartbeat - INFO - Server heartbeat <3
2023-10-02 19:46:51,594 - octoprint.plugins.octorelay - DEBUG - Received the FileRemoved event having payload: {'storage': 'local', 'path': '3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode', 'name': '3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode', 'type': ['machinecode', 'gcode']}
2023-10-02 19:46:51,646 - octoprint.plugins.octorelay - DEBUG - Received the UpdatedFiles event having payload: {'type': 'printables'}
2023-10-02 19:46:55,132 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.102.71
2023-10-02 19:46:55,137 - octoprint.plugins.octorelay - DEBUG - Received the ClientClosed event having payload: {'remoteAddress': '::ffff:192.168.102.71'}
2023-10-02 19:47:13,425 - octoprint.filemanager.analysis - INFO - Starting analysis of local:3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode
2023-10-02 19:47:13,438 - octoprint.plugins.octorelay - DEBUG - Received the FileAdded event having payload: {'storage': 'local', 'path': '3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode', 'name': '3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode', 'type': ['machinecode', 'gcode'], 'operation': 'add'}
2023-10-02 19:47:13,457 - octoprint.filemanager.analysis - INFO - Invoking analysis command: /home/pi/oprint/bin/python3 -m octoprint analysis gcode --speed-x=6000 --speed-y=6000 --max-t=10 --throttle=0.0 --throttle-lines=100 --bed-z=0.0 /home/pi/.octoprint/uploads/3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode
2023-10-02 19:47:15,124 - octoprint.server.util.flask - INFO - Passively logging in user florian from ::ffff:192.168.102.71
2023-10-02 19:47:15,125 - octoprint.access.users - INFO - Logged in user: florian
2023-10-02 19:47:15,469 - octoprint.plugins.octorelay - DEBUG - Received the UpdatedFiles event having payload: {'type': 'printables'}
2023-10-02 19:47:15,523 - octoprint.plugins.octorelay - DEBUG - Received the MetadataAnalysisStarted event having payload: {'name': '3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode', 'path': '3V 5-9_0.8n_0.55mm_PETG_MK3S_5h33m.gcode', 'origin': 'local', 'type': 'gcode'}
2023-10-02 19:47:15,662 - octoprint.plugins.connectandprint - INFO - Uploaded file to print detected, connecting and printing...
2023-10-02 19:47:17,520 - octoprint.util.comm - INFO - Changing monitoring state from "Offline" to "Opening serial connection"
2023-10-02 19:47:17,544 - octoprint.util.comm - INFO - Connecting to port /dev/ttyACM0, baudrate 115200
2023-10-02 19:47:17,546 - octoprint.plugins.connectandprint - INFO - Waiting for printer to connect...
2023-10-02 19:47:17,664 - octoprint.util.comm - INFO - Changing monitoring state from "Opening serial connection" to "Connecting"
2023-10-02 19:47:17,687 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2023-10-02 19:47:18,425 - octoprint.server - INFO - Autorefresh of serial port list stopped
2023-10-02 19:47:21,011 - octoprint.server.util.sockjs - INFO - New connection from client: ::ffff:192.168.102.71
2023-10-02 19:47:21,134 - octoprint.server.util.flask - INFO - Passively logging in user florian from ::ffff:192.168.102.71
2023-10-02 19:47:21,135 - octoprint.access.users - INFO - Logged in user: florian
[... cut ...]
2023-10-02 19:49:27,286 - octoprint.plugins.octorelay - DEBUG - Received the UserLoggedIn event having payload: {'username': 'florian'}
2023-10-02 19:49:27,330 - octoprint.plugins.octorelay - DEBUG - Received the ClientAuthed event having payload: {'username': 'florian', 'remoteAddress': '::ffff:192.168.102.71'}
2023-10-02 19:50:04,789 - octoprint.server.util.sockjs - INFO - Client connection closed: ::ffff:192.168.102.71
2023-10-02 19:50:04,793 - octoprint.plugins.octorelay - DEBUG - Received the ClientClosed event having payload: {'remoteAddress': '::ffff:192.168.102.71'}

```

CrNMGuy, 2023-10-02:
Ok, my bad, Connect And Print (1.2.0) cause the issue

closed 2023-10-02 by CrNMGuy

RobinTail, 2023-10-02:
Yeah, it looks like that the file analysis postpones any other events and code execution besides the "Connect and Print" plugin. I presume it's written synchronously, that's why other threads are waiting.