borisbu/OctoRelay #236: OctoRelay isn't available following the update of Octoprint to 1.9.3

Yoanf26, 2023-10-11:
Hello,

Sorry for my English. I updated Octoprint to 1.9.3 and I no longer have the icons that control my relays via OctoRelay. It’s annoying because I can no longer use my printer 🤦‍♂️

An OctoRelay update is planned, where I have to try to downgrade Octoprint to 1.9.2?

I have some prints planned but that can wait. I have no rush to print my pieces.

Thanks.

RobinTail, 2023-10-12:
Thank you for reporting this, @Yoanf26 .

Are there any error messages in your browser's dev console?
Are there some in OctoPrint log related to OctoPrint?

Yoanf26, 2023-10-12:
First of all, I no longer have the icons at the top right of the web page.

When I try commands via MQTT which was working before updating, I get this type of error

```
RuntimeError: No access to /dev/mem. Try running as root!
2023-10-11 23:56:40,757 - octoprint.plugins.octorelay - INFO - Received the API command getStatus with parameters: {'pin': 'r4', 'command': 'getStatus'}
2023-10-11 23:56:40,795 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin octorelay
Traceback (most recent call last):
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 159, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint_octorelay/__init__.py", line 154, in on_api_command
    return self.handle_get_status_command(data["pin"])
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint_octorelay/__init__.py", line 122, in handle_get_status_command
    is_closed = Relay(
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint_octorelay/driver.py", line 30, in is_closed
    GPIO.setup(self.pin, GPIO.OUT)

```
And when I try a command of this type in ssh, it works

` sudo raspi-gpio set 17 op dh`

RobinTail, 2023-10-12:
@Yoanf26 , thank you for sharing the logs. So, it seems that the error happens when the plugin tries to read the relay state via `RPi.GPIO` library. However, the exact reason is unclear. The CLI command might work because it does not rely on the same python library.
Did you upgrade anything else besides OctoPrint? Did you install any other plugins that operate GPIO?

RobinTail, 2023-10-12:
Similar error was reported in another issue:
https://github.com/borisbu/OctoRelay/issues/35

RobinTail, 2023-10-12:
And please also check this case, @Yoanf26 : 
https://github.com/borisbu/OctoRelay/issues/58#issuecomment-1664368690

It's possible that another OctoPrint plugin is also using RPi.GPIO library but with another type of pins enumeration.

Yoanf26, 2023-10-12:
> @Yoanf26 , thank you for sharing the logs. So, it seems that the error happens when the plugin tries to read the relay state via `RPi.GPIO` library. However, the exact reason is unclear. The CLI command might work because it does not rely on the same python library. [I see it's written in C](https://github.com/RPi-Distro/raspi-gpio). Did you upgrade anything else besides OctoPrint? Did you install any other plugins that operate GPIO?

I don't have other plugins that opérate GPIO. But I have a python script for temperature and humidity sensor, it's works perfect. I don't think he's the culprit because in 1.9.2, everything worked perfectly.

> Could you please also check the version of `RPi.GPIO`, @Yoanf26 ?
> 
> Try a command similar to this:
> 
> ```
> > pip3 show RPi.GPIO
> Name: RPi.GPIO
> Version: 0.7.0
> ```
> 
> Mine is 0.7.0, which is [not the latest one](https://pypi.org/project/RPi.GPIO/) (0.7.1).

I try this command

`Name: RPi.GPIO
Version: 0.7.0
Summary: A module to control Raspberry Pi GPIO channels
Home-page: http://sourceforge.net/projects/raspberry-gpio-python/
Author: Ben Croston
Author-email: ben@croston.org
License: MIT
Location: /usr/lib/python3/dist-packages
Requires:
Required-by: Adafruit-Blinka
`

My script python used Board for use I2C sensor. But even if I disable it and restart the RPI, I still don't have my icons on the octoprint web page. And everything worked in 1.9.2

RobinTail, 2023-10-13:
Ok. @Yoanf26 , could you please also set the DEBUG level of logging for OctoRelay plugin? This should be configurable in Logging section of the OctoPrint settings. I'm hoping that could probably register a little bit more details on the error happening while executing GPIO statement.

Yoanf26, 2023-10-13:
Ok, 

One question, Octoreleys need root ? 

`2023-10-13 18:57:43,900 - octoprint.plugins.octorelay - INFO - Received the API command getStatus with parameters: {'pin': 'r3', 'command': 'getStatus'}
2023-10-13 18:57:43,900 - octoprint.plugins.octorelay - DEBUG - Getting the relay r3 state
2023-10-13 18:57:43,937 - octoprint.server.api - ERROR - Error while executing SimpleApiPlugin octorelay
Traceback (most recent call last):
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint/server/api/__init__.py", line 159, in pluginCommand
    response = api_plugin.on_api_command(command, data)
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint/util/__init__.py", line 1686, in wrapper
    return f(*args, **kwargs)
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint_octorelay/__init__.py", line 154, in on_api_command
    return self.handle_get_status_command(data["pin"])
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint_octorelay/__init__.py", line 122, in handle_get_status_command
    is_closed = Relay(
  File "/home/Yoanf26/oprint/lib/python3.9/site-packages/octoprint_octorelay/driver.py", line 30, in is_closed
    GPIO.setup(self.pin, GPIO.OUT)
RuntimeError: No access to /dev/mem.  Try running as root!
`

See the last line...

Yoanf26, 2023-10-15:
Hello

I found my problem. It's my µSD that's the problem. I wanted to turn off the RPi and it wouldn't. I unplugged it, and Octoprint started in safe mode. I did a reinstallation and Octorelay works.

Sorry for the inconvenience, but until now, nothing indicated to me that the µSD was causing me problems. I am going to order a new one, and replace it, because this one has already been in operation for a few years and especially in devices other than the RPi.

RobinTail, 2023-10-15:
> `RuntimeError: No access to /dev/mem.  Try running as root! `

wow, that's unusual.

> I found my problem. It's my µSD that's the problem

That's great that you figured that out.

> Sorry for the inconvenience

No worries, such issues are very hard to debug 👍🏽 
You did it great. Thank you very much for letting me know the results of your findings.
It may turn out to be helpful for others having similar issue, @Yoanf26

closed 2023-10-15 by RobinTail

appleimperio, 2024-06-10:
@Yoanf26 Can you share how you manage to make it work with MQTT? I'm trying to control a fan from Home Assistant. Thank you

Yoanf26, 2024-06-10:
@appleimperio I stopped using MQTT because it wasn't stable. Errors in reporting information or even orders that don't go through even half the time. And it doesn't work when the 3D printer is not turned on, because Octoprint blocks GCODE commands since it does not detect the printer.

Now I use RestAPI directly on Home Assistant and I created binary sensors and switches in my home assistant configuration.yaml. And there, everything is ok.

I'm sharing with you the configuration for an OctoRelay relay, you just have to change the rx relay number and the choice of name XX of the relay.

binary_sensor:
  - platform: rest
    resource: http://<Host of your Octopi>/api/plugin/octorelay
    method: POST
    headers:
      Authorization: "Bearer <Your API KEY>"
      content-type: "application/json"
    payload: '{ "command": "getStatus", "subject": "rx" }'
    name: Relay XX
    value_template: "{{value_json.status}}"
    unique_id: relay_xx
    scan_interval: 1
    device_class: running

switch:
  - platform: rest
    name: Switch Relay XX
    unique_id: switch_relay_xx
    resource: http://<Host of your Octopi>/api/plugin/octorelay
    body_on: '{ "command": "update", "subject": "rx", "target": "true"}'
    body_off: '{ "command": "update", "subject": "rx", "target": "false"}'
    is_on_template: "{{ is_state('binary_sensor.relay_xx', 'on') }}"
    scan_interval: 1
    headers:
      Content-Type: application/json
      Authorization: "Bearer <Your API KEY>"

To turn the printer on and off I use PSU Control (sorry @RobinTail ) which directly accepts MQTT commands and in the PSU Control configuration it is able to share the switch and the state by the mqtt protocol, even with the printer off. It's perfectly stable.
This is not the simplest method I use but it has worked for several months on my installation. I would have liked to find something simpler, but to tell the truth, since it works, I don't dare touch it anymore. It is true that if MQTT communication would be included in OctoRelay, it would be much simpler.

appleimperio, 2024-06-11:
@Yoanf26 Thank you this work perfectly!