borisbu/OctoRelay #255: Buttons not showing in the navbar

ionicaalexandru, 2024-02-04:
Hi,

did someone encounter this issue: octorelay buttons not showing in the navbar?
Im using octoprint 1.9.3 Python 3.11.2 on a Raspberry Pi 5 8Gb Board with default plugins installed

TNX

ckiefer, 2024-02-06:
Indeed. I have the same setup and I don't see the buttons either.

BroncoNS, 2024-02-10:
Exact same for me as well.

ionicaalexandru, 2024-02-12:
> @ionicaalexandru , @ckiefer , @BroncoNS ,
> 
> > README.md
> > After installing the plugin you need to configure it in order to see the control buttons in the navigation bar.
> 
> Did you configure the plugin and set some of the buttons "Active"?
> 
> ![image](https://private-user-images.githubusercontent.com/13189514/304172993-a7046982-c13a-4e94-ab65-34925eb7efdf.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDc3NjE2ODgsIm5iZiI6MTcwNzc2MTM4OCwicGF0aCI6Ii8xMzE4OTUxNC8zMDQxNzI5OTMtYTcwNDY5ODItYzEzYS00ZTk0LWFiNjUtMzQ5MjVlYjdlZmRmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMTIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjEyVDE4MDk0OFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTcwNWY4MGU0ZDI1OTE2YWZhNjZjMGYzZDMyYzNiMmM4ZDViZmZiM2MyYmFjOWVlNGI2NTI3M2U1MDdlZGZjNmYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.Qw39f5Nfjeiqv7pn2H9JYvr6UuXuhG68kwN_H5cglzM)

yes, there are active buttons for light and fan

RobinTail, 2024-02-12:
Phishing.

ckiefer, 2024-02-12:
Yes, active buttons are defined for air filtration.

RobinTail, 2024-02-12:
1. Please check if there are any errors in the browser console?
2. Please enabled `debug` level of the plugin logging in the Logging section of the OctoPrint, then reload the UI and download the "octoprint.log". In the end of it should be the information prefixed with "The UI feed", having the actual buttons sent to the UI. What do you see there?

ckiefer, 2024-02-16:
@ionicaalexandru , @BroncoNS , @hcooper , @patrickcollins12 , @RobinTail

This is section you are looking for:

`
The UI feed: {'r1': {'relay_pin': 18, 'inverted_output': True, 'relay_state': False, 'label_text': 'Air Filtration', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r2': {'relay_pin': 23, 'inverted_output': True, 'relay_state': False, 'label_text': 'Relay_23', 'active': False, 'icon_html': '[image]', 'confirm_off': False, 'upcoming': None}, 'r3': {'relay_pin': 18, 'inverted_output': True, 'relay_state': False, 'label_text': 'Air Filtration', 'active': False, 'icon_html': '[image]', 'confirm_off': False, 'upcoming': None}, 'r4': {'relay_pin': 23, 'inverted_output': True, 'relay_state': False, 'label_text': 'Webcam', 'active': False, 'icon_html': '[image]', 'confirm_off': False, 'upcoming': None}, 'r5': {'relay_pin': 24, 'inverted_output': True, 'relay_state': False, 'label_text': 'R5', 'active': False, 'icon_html': 'OFF', 'confirm_off': False, 'upcoming': None}, 'r6': {'relay_pin': 25, 'inverted_output': True, 'relay_state': False, 'label_text': 'R6', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r7': {'relay_pin': 8, 'inverted_output': True, 'relay_state': False, 'label_text': 'R7', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}, 'r8': {'relay_pin': 7, 'inverted_output': True, 'relay_state': False, 'label_text': 'R8', 'active': False, 'icon_html': '<div style="filter: grayscale(90%)">&#128161;</div>', 'confirm_off': False, 'upcoming': None}}
`

2024-02-16 10:56:55,677 - octoprint.plugin - DEBUG - Calling on_event on achievements
2024-02-16 10:56:55,677 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on announcements
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on file_check
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on tracking
2024-02-16 10:56:55,678 - octoprint.plugin - DEBUG - Calling on_event on DisplayLayerProgress
2024-02-16 10:56:55,680 - octoprint.plugin - DEBUG - Calling on_event on arc_welder
2024-02-16 10:56:55,681 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2024-02-16 10:56:55,681 - octoprint.plugin - DEBUG - Calling on_event on octoeverywhere
2024-02-16 10:56:55,681 - octoprint.plugin - DEBUG - Calling on_event on octolapse
2024-02-16 10:56:55,682 - octoprint.plugin - DEBUG - Calling on_event on octorelay
2024-02-16 10:56:55,682 - octoprint.plugins.octorelay - DEBUG - Received the SettingsUpdated event having payload: {'config_hash': 'ea196fbb1b884292bc521e2876c1ff26', 'effective_hash': 'c9fba6bb3c32011b4fd6d95fb703a78f'}
2024-02-16 10:56:55,683 - octoprint.plugin - DEBUG - Calling on_event on tplinksmartplug
2024-02-16 10:56:55,809 - octoprint.plugin - DEBUG - Calling get_webcam_configurations on classicwebcam
2024-02-16 10:56:55,813 - octoprint.plugin - DEBUG - Calling get_webcam_configurations on classicwebcam
2024-02-16 10:56:55,816 - octoprint.plugin - DEBUG - Calling get_webcam_configurations on classicwebcam
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on achievements
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on action_command_notification
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on action_command_prompt
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on announcements
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on file_check
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on firmware_check
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on pluginmanager
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on softwareupdate
2024-02-16 10:56:57,542 - octoprint.plugin - DEBUG - Calling on_event on tracking
2024-02-16 10:56:57,543 - octoprint.plugin - DEBUG - Calling on_event on DisplayLayerProgress
2024-02-16 10:56:57,543 - octoprint.plugin - DEBUG - Calling on_event on arc_welder
2024-02-16 10:56:57,543 - octoprint.plugin - DEBUG - Calling on_event on bedlevelvisualizer
2024-02-16 10:56:57,543 - octoprint.plugin - DEBUG - Calling on_event on octoeverywhere
2024-02-16 10:56:57,543 - octoprint.plugin - DEBUG - Calling on_event on octolapse
2024-02-16 10:56:57,543 - octoprint.plugin - DEBUG - Calling on_event on octorelay
2024-02-16 10:56:57,543 - octoprint.plugins.octorelay - DEBUG - Received the DisplayLayerProgress_timerTrigger event having payload: {'updateReason': 'timerTrigger', 'totalLayer': '-', 'currentLayer': '-', 'lastLayerDuration': '-', 'lastLayerDurationInSeconds': '-', 'averageLayerDuration': '-', 'averageLayerDurationInSeconds': '-', 'currentHeight': '-', 'currentHeightFormatted': '-', 'totalHeight': '-', 'totalHeightFormatted': '-', 'feedrate': '-', 'feedrateG0': '-', 'feedrateG1': '-', 'fanspeed': '-', 'progre [... cut ...]
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/__init__.py", line 320, in update_ui
    relay_state = relay.is_closed() if active else False
                  ^^^^^^^^^^^^^^^^^
  File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 30, in is_closed
    GPIO.setup(self.pin, GPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address

I am running on
- https://github.com/OctoPrint/OctoPrint/releases/tag/1.10.0rc2
- https://github.com/borisbu/OctoRelay/releases/tag/4.0.1

RobinTail, 2024-02-18:
Thank you, @ckiefer 

First thing I noticed is that the first feed reports all relays inactive, which makes their control buttons invisible.

[image]

Then I see you're setting the third relay active

```
2024-02-16 10:57:42,298 - octoprint.plugins.octorelay - INFO - Saving the settings: {'r3': {'active': True}}
```

But it fails due to an error, which is caused by GPIO driver

```
File "/home/pi/OctoPrint/lib/python3.11/site-packages/octoprint_octorelay/driver.py", line 30, in is_closed
GPIO.setup(self.pin, GPIO.OUT)
RuntimeError: Cannot determine SOC peripheral base address
```

Which is unusual and may be an exclusive issue of Raspberry Pi 5.
Here I found some information on it:

https://www.reddit.com/r/RASPBERRY_PI_PROJECTS/comments/188yynn/im_having_soc_problems_in_rpigpio_library_can_any/

What is also worth to notice:

> I am running on
> https://github.com/OctoPrint/OctoPrint/releases/tag/1.10.0rc2

that OctoRelay is not yet completely tested on on OctoPrint 1.10.x.

RobinTail, 2024-02-18:
Here is the solution that reported to solve the issue:

https://www.reddit.com/r/RASPBERRY_PI_PROJECTS/comments/188yynn/comment/kbqbc2w/?utm_source=share&utm_medium=web3x&utm_name=web3xcss&utm_term=1&utm_content=share_button

> it looks like it was a 64-bit bug with addressing that did not have a case in the Python Rpi-gpio library for that OS version on Pi5. Using the 32-bit version of Rasberry Pi OS was able to resolve the error and make it work as expected.

@ckiefer , @ionicaalexandru , @BroncoNS

RobinTail, 2024-02-18:
Probable related issue: https://sourceforge.net/p/raspberry-gpio-python/tickets/214/

ionicaalexandru, 2024-02-25:
For me 32 bit os did not resolve the issue

aaronmussig, 2024-07-04:
Hi, I forked this and used `pinctrl` to control the GPIO pins. Since it's not backwards compatible I haven't made a PR, but if you want to use it, you can use this release: https://github.com/aaronmussig/OctoRelay/releases/download/v1.0.13/release.zip

closed 2024-07-10 by RobinTail

RobinTail, 2024-07-10:
@ionicaalexandru , @ckiefer , @BroncoNS and @aaronmussig ,

I'm glad to announce that thanks to the contribution of @ch3p4ll3 I managed to release version 5.0.0 that supports Raspberry Pi 5

PR #280 (2024-07-10): 🚀 5.0.0
Closes #255 
Closes #212 

- #274 
- #281 
- #215 
- #282 
- #283