borisbu/OctoRelay #259: Specific Gcode commands for turning relays on or off

OdinKadaver05, 2024-02-18:
Currently the ony Gcode command available is for switching back and forth. This works in most cases, but in my opinion there are also many cases where it would be easier to just have a specific command for on and a specific command for off, instead of having to expect a specific state of the relais before executing the Gcode command. 

Example: I have been building an enclosure for my printer in the past few month which also uses buttons to execute commands and LEDs to give feedback. On of these buttons is for Play/Pause, which is supposed to be off when nothing is being printed, on while printing and blinking when a print has been paused for some reason. I had to use 2 relais and a blinker circuit to make this work, but if I stop the print while it's paused I have no way of making the second relais switch back to the default state, because there is no specific command for turning it off, only for switching it back and forth.

RobinTail, 2024-02-18:
I agree

RobinTail, 2024-02-18:
If I make this:

```
@OCTORELAY  r#  [ON | OFF]
```

would it fulfil the need, @OdinKadaver05 ?

OdinKadaver05, 2024-02-19:
@RobinTail 

Definitely. Thanks for the fast response 👍 😃

closed 2024-02-19 by RobinTail

RobinTail, 2024-02-19:
🚀 The feature released in version 4.1.0, @OdinKadaver05