borisbu/OctoRelay #263: A way to use physical buttons

OdinKadaver05, 2024-04-03:
I'm currently trying to make build an enclosure for my MK3s which also utilizes physical buttons for power etc. . Since the option of using physical buttons isn't integrated into OctoRelay I use the Physical Buttons add-on to execute Gcode-commands which then are picked up by Octorelay and then executed by the relais. This seemed to work fine until I noticed that apparently Octoprint doesn't just turn off the Gcode terminal when no printer is connected, it also disables the ability to execute Gcode commands from every other source, which is kindof problematic if I want to use this for a power button that turns off the printer but keeps running.

Long story short: it would be great if either inputs could be directly processed by octorelay or if there would be a simple way to use what physical buttons calls system commands. Some other solution to this

RobinTail, 2024-04-04:
If I understand correctly, you'd like to be able to configure additional GPIO pin for relay so that the plugin could read its state and interpret it somehow as a signal for switching the relay. Does that appear to be correct, @OdinKadaver05 ?

RobinTail, 2024-04-04:
Similar issue (resolved): https://github.com/borisbu/OctoRelay/issues/36
Similar issue (closed): https://github.com/borisbu/OctoRelay/issues/60

RobinTail, 2024-04-04:
Check this out, @OdinKadaver05 : 
https://github.com/borisbu/OctoRelay/issues/36#issuecomment-928911843

Could that approach work for you?

RobinTail, 2024-04-04:
Physical Button plugin mention in that thread:
https://github.com/LuxuSam/PhysicalButton

OdinKadaver05, 2024-04-04:
@RobinTail  Yes and no. In fact I'm already using that Plugin for controlling my fan, the lights and a play/pause button, but my problem is that I also want to control the power of my printer this way, while being able to also control it over the Webinterface and OctoApp.

I've hooked the relay for the PSU up and the automatic stuff (such as "turn relay on when server starts") works fine, but since the only way that I can get the Physical Button Plugin to send commands to OctoRelay is with the Gcode option in Physical Button, this doesn't work unless my printer is already powered on due to octoprint (from what I can tell) not processing Gcode commands while no printer is connected, even if the commands are from Plugin to Plugin.

[image]
[image]

OdinKadaver05, 2024-04-04:
Sry, didn't mean to close the "issue. Wrong button....

RobinTail, 2024-04-04:
> the only way that I can get the Physical Button Plugin to send commands to OctoRelay is with the Gcode

Did you try the "Activity" called "Output" in that plugin?

[image]

When I configure it to the same GPIO that my printer relay is connected to, time=0 and "Active"="Toggle", I can confirm that I can turn my printer on with a physical button.

Check out my setup (made it tonight) — it's not so pretty though, but I wanted to check how it works, @OdinKadaver05 :

https://github.com/borisbu/OctoRelay/assets/13189514/c8b13cd0-9fbe-4ed8-9553-68b4506a2e26

OdinKadaver05, 2024-04-04:
Thank you so much. I can't believe it took me so long to find out that this works......

I originally planned to use a plugin called enclosure plugin, but unfortunately this has been abandoned (I guess) by now and doesn't work the way it's supposed to, which is when I found tho solution of combining Physical buttons with OctoRelay. I guess I must have missed this when I was setting it up since it was rather late at night.......

The setup looks good for just one evening of work (I'm fairly certain that I couldn't have gotten it to work so fast). I have been working on mine since almost 2 years by now (if you include the planning in fusion 360). Its main purpose was supposed to be that I can sleep while having the printer run at night and I just kept on having more and more ideas on what to add (air filtration, the control panel, multiple cameras, Error detection with Obico, ....). Admittedly there were phases where I did more or less depending on the time I had but now it finally seems like a "finish-line" is in sight.

Once more thanks for the tip that I missed this. Knowing that I can also do it this way makes things som mus easier.

closed 2024-04-04 by OdinKadaver05

RobinTail, 2024-07-21:
In version 5.x of the plugin, due to the new driver being used, the Physical Button can not operate same pins that are already taken by OctoRelay.
However, you can still use System Command Action to operate relays from that plugin:
https://github.com/borisbu/OctoRelay?tab=readme-ov-file#change-the-relay-state

OdinKadaver05, 2024-09-19:
@RobinTail Is there a way to make it toggle the current state? I could only find the commands for either turning it on or turning it off.

RobinTail, 2024-10-12:
There is probably a way to bypass the lock (exclusive pin reservation) established by `gpiozero` with a single line, that I'm currently investigating here, trying to make a cleanest solution possible:
https://github.com/borisbu/OctoRelay/pull/315

RobinTail, 2024-10-26:
The Physical Button plugin compatibility is about to be fixed in v5.1.0 of OctoRelay.

PR #321 (2024-10-26): 🚀 5.1.0
Fixes #263 
Fixes #299 

- #320 
- #322 
- #315